IHP-GmbH/IHP-Open-PDK #2: Invers operation correct?

dwarning, 2023-04-04:
Hello,
I made few check with the vbic's in ngspice. Most things are plausible, but invers operation seems a bit strange. That's said without knowledge of the npn13G2 structure/crossview.
Can you comment the attached picture made with Vce=-1V and Vbc=0.3 ... 1V.
Thank you for the Open-PDK activity.
[image]

bero4sc, 2023-04-08:
It is a HBT device. Why did you reverse bias the collector to emitter?

dwarning, 2023-04-08:
Ok, answer by myself. Model seems plausible, if I see the lower Early voltage in reverse operation compared to forward.
BTW - at the moment the vbic temp control is buggy in ngspice - will be repaired in next release.
[image]

closed 2023-04-08 by dwarning