IHP-GmbH/IHP-Open-PDK #3: missing model for sg13_lv_nmos

proppy, 2023-05-31:


proppy, 2023-05-31:
Is that expected? (i.e: the model for nmos/pmos will only get developed/released later)

sergeiandreyev, 2023-05-31:
Hi! Yes, this is expected.. For now, we have only models for HBT (BJT) devices, for the MOSFET models we're still discussing the Spectre to SPICE format translation with 3rd parties. I hope they will be released in summer timeframe.
Also we're considering to put nmos/pmos Spectre models to the GitHub (but this will certainly help you only in case you have access to the simulator that supports them)

proppy, 2023-05-31:
maybe @RTimothyEdwards and @atorkmabrains can share some insight on how the conversation was done respectively for SKY130 and GF180MCU.

atorkmabrains, 2023-05-31:
@sergeiandreyev @proppy we actually use home grown scripts for this. But we also go into debugging the model card parameters and tweak them to match the desired outcome. That's why we always request a golden measurements data that we could use as a reference. It's very lengthy process if we faced issues related to not matching behaviors. 

@sergeiandreyev Please contact me privately at linkedin if you want us to collaborate: https://www.linkedin.com/in/amro-t-24701458/

atorkmabrains, 2023-05-31:
@sergeiandreyev You don't need to release spectre Model cards for that BTW.

mithro, 2023-05-31:
https://semimod.de/ -- who is based in Europe might also be able to offer assistance, particularly if you are doing recharacterization stuff or regenerating from raw data.

The topic of spice model generation and validation comes up at lot in the CHIPS Alliance Analog work group (https://www.chipsalliance.org/workgroups/) and on the SKY130 raw data repository (https://github.com/google/skywater-pdk-sky130-raw-data).

metroid120, 2023-06-01:
Markus from SemiMod: Thanks for your efforts @noherbrferurtth @proppy 
We are looking forward to helping with the model conversion and will join the IHP meeting that is coming up.

Also, I am delighted to share that we recently got funding for developing an open-source MOS Model extraction tool, so we are currently looking for (i) reference models and (ii) a test chip for measurement. 
I think the IHP technology would be perfectly suited.

sergeiandreyev, 2023-09-13:
- we have now MOS Spectre models available in the repo: `libs.tech/spectre`
- and SemiMod @metroid120 did an outstanding job to convert these models for use w/ ngspice & Xyce (also HBT models were updated)
   The data is located in their GitLab repository https://gitlab.com/dmt-development/ihp_sg13g2_compact_models

sergeiandreyev, 2023-10-24:
update: we now have device simulation models available in the repo: `lib.tech/ngspice/models`
for usage please consult respective readme files and documentation on SemiMod GitLab [repo](https://gitlab.com/dmt-development/ihp_sg13g2_compact_models)

sergeiandreyev, 2024-03-23:
Hi @proppy , I guess this issue is resolved now.. Can we proceed and close this item as  fixed?

KrzysztofHerman, 2024-06-04:
Mosfets models available in `libs.tech/ngspice/`

closed 2024-06-04 by KrzysztofHerman