IHP-GmbH/IHP-Open-PDK #9: diode devices DRC and models

stafverhaegen-chipflow, 2023-08-31:
In the layout rules manual there is mention of a `Recog.diode` layer and this layer is also used in the `sg13g2_pr.gds` in for example the `dpantenna` cell.  
No devices for diodes using this layer are present in the layout rules or the process specification manual. Neither could I find models for diodes.

sergeiandreyev, 2023-09-14:
There are no design rules related to `Recog.diode` directly because it is only a recognition layer w/o any physical meaning. And the `dantenna` and `dpantenna` devices are not present anywhere in DRM because there are no any design rules specific for these cells (whereas, for example, Schottky diode has specific rules and is described in the DRM).
Please let me know if that answers your *first* question.
For the models - we are going to convert them for the open source simulator support and then upload to our repository.

Regards,
 Sergei

FatsieFS, 2023-09-14:
Once the models are there I do think it makes sense to have a section on them in the `Device layout rules` Chapter and have rules for `Recog.diode` like `Rsil.c` for `RES`.
I suppose `Recog.diode` does play a role in LVS.

sergeiandreyev, 2024-03-23:
Hi @FatsieFS , now we have diode models in place available in our OpenPDK repository. I'll let you know regarding the update of the documentation.

closed 2024-06-04 by KrzysztofHerman