IHP-GmbH/IHP-Open-PDK #10: ihp pdk for ADS

yuhan-tan, 2023-09-03:
Hi experts,

Sorry if this is the wrong place to comment, I didn't find any other portal for this issue.

I am currently doing my mmWave PA design and wanna using IHP PDK for ADS, however, I tried a lot to load the models in ADS and failed. every time loading the netlist/spectre file [sg13g2_moslv_psp_parm.scs](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.tech/spectre/sg13g2_moslv_psp_parm.scs) in libs.tech throws an error of syntax error. I am wondering if this is the problem of open PDK or from keysight ADS. 

Hoping to receive your reply. I really appreciate any help you can provide.

Best,
Yuhan

sergeiandreyev, 2024-03-23:
Hi @yuhan-tan , as of today we have the simulation models as part of our OpenPDK repository, also we have initial Qucs-s tool support. Could you try if with the latest update of the PDK (`dev` branch) you can work on the required tasks?

KrzysztofHerman, 2024-06-04:
@yuhan-tan do you have any more questions or doubts or can we close the issue ?

closed 2024-07-10 by sergeiandreyev