IHP-GmbH/IHP-Open-PDK #20: Xschem example circuits should not contain `pre_osdi`

hpretl, 2023-11-16:
In situations where the VerilogA models are located in a central location (not in various simulation directories, but e.g. in `.../ngspice/lib/ngspice`) the `pre_osdi` directive leads to an error when starting `ngspice`.

As a consequence, this directive should not be part of the "standard" examples. Alternatively, it should be wrapped into a check whether this file exists in this local directory.

olisnr, 2023-11-21:
i think the `pre_osdi ./psp103_nqs.osdi` belongs to the

`https://gitlab.com/dmt-development/ihp_sg13g2_compact_models`

hpretl, 2023-11-21:
Yes, the statement `pre_osdi` loads the external PSP model into ngspice which is needed for the IHP MOSFET. But this model can be loaded from a central location with a central ngspice setting. Thus, the `pre_osdi` is not needed in every schematic.

olisnr, 2023-11-21:
You would prefer `./simulation/.spiceinit` then?

hpretl, 2023-11-21:
Not really. In our Docker image we just copy `psp103.osdi` into `.../ngspice/lib/ngspice`, and uncomment the corresponding line in `.../ngspice/share/ngspice/scripts/spinit`:

```
* Load the OpenVAF/OSDI models
if $?osdi_enabled

 * osdi /foss/tools/ngspice/ngspice/lib/ngspice/asmhemt.osdi
 * osdi /foss/tools/ngspice/ngspice/lib/ngspice/bjt504t.osdi
 * osdi /foss/tools/ngspice/ngspice/lib/ngspice/BSIMBULK107.osdi
 * osdi /foss/tools/ngspice/ngspice/lib/ngspice/BSIMCMG.osdi
 * osdi /foss/tools/ngspice/ngspice/lib/ngspice/HICUMl0-2.0.osdi
 osdi /foss/tools/ngspice/ngspice/lib/ngspice/psp103.osdi
 * osdi /foss/tools/ngspice/ngspice/lib/ngspice/r2_cmc.osdi
 * osdi /foss/tools/ngspice/ngspice/lib/ngspice/vbic_4T_et_cf.osdi

end
```

This `spinit` is the default init `ngspice` reads.

hpretl, 2023-11-21:
Since quite a few `.osdi` files are already there, and are included in the same way, it feels natural to add the PSP model there as well.

hpretl, 2024-07-17:
It looks like this will be resolved soon: https://github.com/IHP-GmbH/IHP-Open-PDK/pull/162

closed 2024-08-20 by KrzysztofHerman