IHP-GmbH/IHP-Open-PDK #22: question is the "sg13g2_pr.gds pnpMPA" usefull for band-gap references?

olisnr, 2023-11-21:
or what is it for?

olisnr, 2023-11-22:
and what is the "ng"-parameter in the MOS-models? is it "number of gate-strips"? and how does it compare to the "m"-parameter?

sergeiandreyev, 2023-11-23:
You're right about `pnpMPA` - this device is a pnp HBT and it is used for band-gap reference circuit.

sergeiandreyev, 2023-11-23:
`ng` stands for 'Number of gates' - Forms a multi-finger gate. The individual channel width will be the total width divided by `ng`.

sergeiandreyev, 2023-11-23:
and `m` is for 'number of parallel instances'. I'll see if I can provide some clear explanation how these two parameters (if) are interconnected

olisnr, 2023-11-23:
thanks a lot. 

then i have some other questions
- is there a spice-model for the pnpMPA, or should i use some BC560B as model?
- is dantenna and dpantenna thought as gate-protection, against static charge in production?
- what is ntap and ptap? some bulk-connection?
- what is the sealring good for?

sergeiandreyev, 2023-11-23:
you can try to play with the nmos Pcell in KLayout that we have as part of our repo `libs.tech/klayout` to see how `ng` and `m` are working

olisnr, 2023-11-23:
> you can try to play with the nmos Pcell in KLayout that we have as part of our repo `libs.tech/klayout` to see how `ng` and `m` are working

thanks: `ng` changes the shape of the transistor but not `W` and `L`. `m` does nothing with the pcell and if i see this correct, the layouter has to place and route `m` times this pcell in parallel. agree?

closed 2023-11-29 by olisnr