IHP-GmbH/IHP-Open-PDK #24: Simulation problem with diode model and elevated temperatures

stafverhaegen-chipflow, 2023-11-27:
I have a test case of an IO cell I am working on: [hightemp_sim.zip](https://github.com/IHP-GmbH/IHP-Open-PDK/files/13476981/hightemp_sim.zip)
You can run the simulations with the `./sim.sh` and that generates the included `sim25c.out` and `sim70c.out` files. As you can see the simulation works for 25℃ but not for 70℃. For 70℃ the reporter error is:
```
doAnalyses: TRAN:  Timestep too small; initial timepoint: trouble with darea-instance d.xtop.xnclamp.xdgate.d1
```

KrzysztofHerman, 2024-02-26:
I have consulted this unusual behavior of the simulation with the developer of ngspice prof. Holger Vogt. He noticed that the model of the permi diode, which conforms the dantenna diode has an unusually low value of $v_j$ parameter, what causes convergence issues. The parameter itself varies with the temperature, what explain the situation that at $T=25$ the simulation converges and at $T=70$ does not. We will update the model as soon as we derive a valid value.

sergeiandreyev, 2024-02-29:
we see the warnings in commercial simulator as well, but the default behavior we have there is that the `vj` parameter value is constrained to a larger value, this is why the simulation is not failing for internal PDK

sergeiandreyev, 2024-03-01:
diode model `dperim` was fixed

sergeiandreyev, 2024-03-23:
Hi @stafverhaegen-chipflow , could you please confirm that the issue is resolved and can be closed. 
Thanks,
 Sergei

stafverhaegen-chipflow, 2024-03-25:
I can confirm that the simulations now run at elevated temperature.
Sorry for the delay.

closed 2024-03-25 by stafverhaegen-chipflow