IHP-GmbH/IHP-Open-PDK #25: sg13_lv_nmos ignores the "m" parameter

olisnr, 2023-11-28:
the **sg13_lv_nmos** model should accept an `m` parameter in spice-simulation. at least the Xschem-symbol has this parameter and any MOS-device should, because its a spice-standard. but i ngspice-simulations this multiplier is ignored. actual i would expect something like multiplying the parameter W.
in the file `IHP-Open-PDK-main/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_moslv_mod.lib` there is a parameter `mfact=1`. may be thats the `m`?
but `XM1 net3 net1 GND VSS sg13_lv_nmos W=2.0u L=0.45u ng=1 mfact=100` also dosnt work...

metroid120, 2023-11-28:
Hi, the model description is given here:
https://gitlab.com/dmt-development/ihp_sg13g2_compact_models
Maybe we should put a description also somewhere in this PDK?

The parameter "ng" should do what you want. The "M" parameter or "MFACT" for PSP model was not implemented/used here.

olisnr, 2023-11-28:
the ng-parameter makes the same W via many smaller W/ng. if i make an 10u/1u ng=5 FET, then it produces a fet with 5 fingers and about W=10u (per FET not per finger).

metroid120, 2023-11-28:
Yes, I agree this seems odd. Can you post a Screenshot of the FET in Klayout with ng=2?

miesli, 2023-11-28:
Hi,

as @metroid120 already stated, we followed along the original spectre model, because this was verified against measurement by IHP. Why it was used like this, we do not know. There are multiple possible explanations, but this would be speculative. Our goal was to have the OpenPDK as a drop-in replacement for SG13G2.

On the other hand: In the PSP Verilog-A code, there are instance parameters called `mult` (number of devices in parallel),  `ng` (number of gate contacts) and `nf` (number of fingers).  And in the code you can see, that `mult` is multiplied to the currents and charges at the output. So using `mult` would be the way to go. 

So, one possible step is to do a netlist extraction from different FET layouts to see what kind of netlists are created. Is this already possible in the OpenPDK?

olisnr, 2023-11-28:
`Nsg13_lv_nmos d g s b sg13g2_lv_nmos_psp w='w/ng' l=l mult=ng` mult is already used inside the model. its not accessible from spice, maybe i can change the model to   `Nsg13_lv_nmos d g s b sg13g2_lv_nmos_psp w='w/ng' l=l mult=ng*m` ?

olisnr, 2023-11-28:
if i understand the `m` and `mult` parameter correct, its about to be able to use some parts in parallel, without the need of drawing all this parts and wire it together. 

its clear that a real circuit has real wires, not ideal ones like in the schematics. but the factor `m` and `mult` dont have to do something other then connect many parts without the real effects of wiring. this effects will be visible anyway only at time of parameter-extraction.

olisnr, 2023-11-28:
the idea with changing the netlist would work: i could use TCL in Xschem to generate `m` times the same MOSFET in the spice-netlist. but then the sad ngspice has to simulate all this parts. a `m` or `multi` factor will reduce the simulation time great if You need many parts parallel...

olisnr, 2023-11-28:
if i change all the models to         `Nsg13_lv_nmos d g s b sg13g2_lv_nmos_psp w='w/ng' l=l mult='m*ng'` **mult='m*ng'** instead of **mult=ng**, then m does what i would expect:

[image]

metroid120, 2023-11-28:
only at first glance, this is correct.
In the original specter model call

```
        sg13_hv_nmos (d g s b) sg13g2_hv_nmos_psp w=w/ng l=l m=ng   
        + as=max(w/ng,wmin)*(z1+((ng-1)/2)*z2)/ng   
        + ad=max(w/ng,wmin)*(z1+((ng-1)/2)*z2)/ng  
        + ps=2*(max(w/ng,wmin)*((ng-1)/2+1)+z1+(ng-1)/2*z2)/ng  
        + pd=2*(max(w/ng,wmin)*((ng-1)/2+1)+z1+(ng-1)/2*z2)/ng  
        + trise=trise  
        + ngcon=2 
```

as, pd etc. are scaled according to ng. I assume if you also multiply those by "m" it **_MIGHT_** be correct.

olisnr, 2023-11-28:
im not sure: `m` or `multi` should do this multiplications inside the model. 

but i will try some ac-simulations also and see what comes out of the simulation if i replace ng with 'ng * n' everywhere.

olisnr, 2023-11-28:
and here comes ng=1 and ng=2 from KLayout

[image]

miesli, 2023-11-28:
Can you also show mfact = 1 and = 2?

PS I assume that all the other parameters did not change.

olisnr, 2023-11-28:
this are the settings of the fets in KLayout:
[image]
[image]

miesli, 2023-11-28:
Thank you for showing this. 

ng is then the thing the documentation in the model files says it is. It allows using the same drain twice.

m or mfact is not used neither in the layout nor the circuit models. 

It seems that the SG13G2 PDK does not support MOSFET multiplication. To get more current, you should increase the size or place multiple instances in parallel. Of course, this is impractical and increases simulation time.

It is a good thing that this PDK is open :). I would recommend for the "netlist phase" to go for the mult=m*ng you described above where you also set ng=1. Later on, when you iterate on the layout, go to m=1 with parallel instances and ng=1 or 2 depending on your layout needs.

Also, in favor of having the openPDK as a drop-in replacement for SG13G2, I would not change the "main" model files. But of course that is up to IHP to decide. As promised, I will improve the documentation on our model-repo.

olisnr, 2023-11-28:
yes i see it also this way, but i would be glad, it iHP would adopt the `m` parameter in the spice-models, so that there is no need to try to improve it with trail and error...

sergeiandreyev, 2023-12-11:
we need to understand the root cause of this issue and to discuss this with Semimod guys, I have an internal testcase with the commercial tools/models of our PDK and the `m` parameter is working there (the change in the value => changes the simulation result as expected)
unfortunately, as you understand, I cannot share the full testcase for the open source community

metroid120, 2023-12-12:
At least the "mfact" parameter of the models is completely ignored by the original models. 

I assume that maybe "m" somehow changes "ng" via the PDK integration that we have not seen. Not sure at this point.

dwarning, 2024-02-06:
Out of curiosity: mfact is used in the flatband calculation parameter vfbo.
The actual mix of ng, nf and mult shows that the original spectre lib is not able to describe separate device multiplication.

sergeiandreyev, 2024-03-23:
Hi all, so with the latest update from @miesli  I guess this issue could be closed as fixed, could you please confirm @olisnr ? @KrzysztofHerman checked the behavior against the commercial tools/PDK and now we see identical results, so let's assume we have the models aligned in this regard. 
On the multi-finger device measurements, unfortunately best case we will have them end of this year / begging of next.. So we cannot decide on the quality of internal implementation until then.

PR #58 (2024-03-19): removed the mfact parameter and implemented m properly
Fixes #25 

As mentioned in the discussion, this fix is not ideal, but it seems this is how the original spectre SG13G2 model extractors did it. One main open question is the parameter `m`, which is used in the scaling of the PSP model parameter `vfbo`.

This can not be verified at the moment since in the current available measurements there are no multi-finger device measurements present.
closed 2024-06-04 by KrzysztofHerman