IHP-GmbH/IHP-Open-PDK #34: Unexpected behavior/values seen in small-signal models of thin-oxide devices.

bmurmann, 2023-12-10:
1. The gate overlap capacitances seem to be very large. An op analysis for a 5/0.13 NMOS (sg13_lv_nmos) gives cgsol = 3.19575e-15 fF. This corresponds to 0.64 fF/um, about twice as large as in other 0.13 um CMOS processes. Are these realistic models?
2. A likely non-physical kink is seen when running a VGS DC sweep and plotting fT = gm/Cgg/2/pi for the NMOS device (sg13_lv_nmos).
[image]
For reference, here is my Jupyter notebook that produces these data:
https://github.com/bmurmann/Ngspice-on-Colab/blob/main/notebooks/IHP_SG13G2_VGS_sweep.ipynb

metroid120, 2023-12-12:
Dear Boris,

find attached our results documentation for the compact models of the MOSLV. We converted the existing spice models to ngspice format.

_**Regarding fT:**_

As you can see, the fT does not show such a kink for the analyzed devices.
The way you plot fT is simple, but only has limited meaning. It is more rigorous to plot 

fT = f/Im(Y11/Y21)   (where f is a spot frequency)

You must run an AC analyis.

[image]

_**Regarding Overlap caps:**_

Let us do an approximate calculation based on the model parameters of the LV NMOS and the model equations.
The total GS overlap charge in the model is 

`                    Qfgs         =  Qfgs + Qgs_ov;`

The model parameter CFRW (outer fringe capacitance for 1 um wide device)

`'((1-rfmode)*2e-16 + rfmode * (1e-18 + pre_layout * (ng>0 ? 3.8525e-17 : 0)))/ng' `

evalutes to 0.2 fF/um for rfmode=1. It sets Qfgs.

The model variable that sets Qgs_ov is

`                        CGOV_p       = `EPSO * EPSROXO_i * WEcv * LOV_i / TOXOVO_i;`

, which depends on several model parameters that are physics-based. We can approximately evaluate this capacitance to be

CGOV_p = 8.85e-12 F/m 3.9*5um*0.13um/2.2nm = 10.2 fF

I ignored some stuff here, like the difference between drawn and actual channel.
The total CGSOV for a 0.13/5 device is 11.2 fF. This corresponds to around 2 fF/um. 
So somehow even higher than what you have. Not sure what to make of this.

[documentation.pdf](https://github.com/IHP-GmbH/IHP-Open-PDK/files/13646005/documentation.pdf)

bmurmann, 2023-12-12:
1) I would say it's the other way around regarding limited meaning. If I use the total cap (including huge overlap caps) and plot against log(ID) I also no longer see that kink; it is a nice way to swamp it out and sweep things under the rug 8). So, this has nothing to do with h-parameters, but with the modeling of the intrinsic transistor.
From the way you responded, I see that you think about fT like a microwave designer. There is another world of low frequency design where we think of the ratio gm/Cgg as a figure of merit that tells us how much gm we get per invested gate cap. In that world, we are not interested in the physical unity gain frequency, and it makes no sense to burden circuit design with h parameters. Instead, what we need is an accurate accounting of gate capacitance at a given gm. Here it also make sense to distinguish intrinsic and extrinsic components for model sanity checks (as I did).
2) You may know that for the longest time of this industry (before Dennard scaling ended), 2fF/um was a fairly technology independent value accounting for the total (intrinsic + extrinsic) gate capacitance of a MOSFET. After Dennard ended, this got closer to 1.5 fF/um. Now, with that in mind, it is obvious that having an overlap capacitance of about 2.2 fF/um (per your calculation) either means the technology or the model is broken. I am not sure how I can help in making that determination. Here is a table of typical parameters from Tony Chan Carusone's textbook:
[image]

 
[image]

sergeiandreyev, 2023-12-13:
Hello Boris,
 I reproduced the large gate capacitance values in your testcase using our commercial tools/PDK. On my side these are:
`cgsol = 3.225496 fF
cgdol = 3.123428 fF`

I'm forwarding this issue to our IHP modeling experts..

metroid120, 2023-12-13:
The physical origin of the large cgsol is largely the LOV parameter (see my parameter) that can be taken from cross-sections. One could check that.

bmurmann, 2023-12-13:
Thank you, everyone. BTW, I looked at the docs here: https://github.com/IHP-GmbH/IHP-Open-PDK/tree/main/ihp-sg13g2/libs.doc/meas/MOS/doc. I am getting very large values for Cov when I divide the measured cap by the give width. But, I am not sure if I am interpreting the data correctly.

bmurmann, 2024-03-20:
A couple more observations for this thread:
1) The process spec says that the nominal "Miller cap" is 0.36fF/um for an NMOS.
[image]
2) For a 10/0.13 NMOS, the simulated cgdol=6fF, while 3.6fF is expected from the process spec. The simulated value changes to 4.6fF with ng=4, which is hard to explain when I think about the actual layout. I can see how some deltaW affects this in a fingered device, but in reality splitting up a device does not really give you smaller caps (due to fringing).

hpretl, 2024-07-17:
See here for an observation regarding the model usage of number of fingers: https://github.com/IHP-GmbH/IHP-Open-PDK/issues/164

dwarning, 2024-10-23:
Try to understand the problem and made few AC simulations with two 10/0.13 lv_nmos devices (NF=1 and 4):

[image]

The simulations show a value of 12fF for both devices which I must divide by 2*10um because of Cgdov and Cgsov in parallel. 

@bmurmann: You said that you simulated and got for a fingered device a much lower value of 4.6fF. How you get this values? Are these OP information's as shown in your initial setup?

bmurmann, 2024-10-23:
I don't have think I have this simulation anymore, which was back in March, and also before some changes were made to the models. I will try to duplicate this early next week when I am back from travel. In any case, if Cgdov is 0.6 fF/um, it does not match the given process spec:
[image]