IHP-GmbH/IHP-Open-PDK #37: it seems like the pnpMPA model and dc_pnpMPA.sym have different pin-orders

olisnr, 2023-12-21:
in the file `IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib` there is a model:
```
.subckt pnpMPA e b c
.param a=2p p=6u ac=13.33p pc=14.64u
+ dev_a=a*1e12 dev_p=p*1e6 sub_a=ac*1e12 sub_p=pc*1e6

QpnpMPA e b c pnpMPA_mod area=dev_a
...
```
but if i run the netlist-generation of Xschem for `IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/xschem/sg13g2_tests/dc_pnpMPA.sch` it generates:
```
** sch_path: IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/xschem/sg13g2_tests/dc_pnpMPA.sch
**.subckt dc_pnpMPA
XQ1 net3 net1 E1 pnpMPA
Ve net2 E1 0
.save i(ve)
Vb net1 GND 0
.save i(vb)
Vc net3 GND 0
.save i(vc)
I0 GND net2 1u
...
```

it seems to me that `c` and `e` are interchanged somewhere.

olisnr, 2023-12-22:
i think the description of the model is wrong and should be `QpnpMPA c b e pnpMPA_mod area=dev_a` because the gain of the transistor is a bit over 1, and thats like in the measurement data, where the gain is typical about 1.5.
and if i flip the pnpMPA in the test-circuit so that `c` is `e` and versa vice, the gain is much lower.

i would also like to know, why the different DUT-measurements of the pnpMPA in `IHP-Open-PDK-dev/ihp-sg13g2/libs.doc/meas/pnpMPA` are that different. i dont see any differences in the headers of `fg_vcb0_d0406_DUT1.mdm .. DUT17.mdm`, its always
```
  TNOM "27"
  TEMP "27"
  TECHNO "SG13S"
  LOT "PQA701"
  WAFER "17"
  CHIP "d0406"
  MODULE "T323_pnpMPA"
```
so i would think, the data should be more or less the same for all DUTs

sergeiandreyev, 2024-01-15:
the comments section has a specific DUT defined for each of the .mdm files, and these are different:
```
  DEV_NAME "DUT4"
```
[image]

sergeiandreyev, 2024-01-15:
on the pin order, I'm checking internally

sergeiandreyev, 2024-07-22:
Hi @olisnr, we uploaded to the repository the measurement documentation for the `pnpMPA` device ([link](https://github.com/IHP-GmbH/IHP-Open-PDK/tree/dev/ihp-sg13g2/libs.doc/meas/pnpMPA/doc)) - could you please check it and give your feedback if there are still some open questions regarding the device
If this is the case maybe we should have a video call then

olisnr, 2024-07-24:
there is also a link-problem in the HBT doc:
`file:///home/ich/share/pdk/ihp-sg13g2/libs.doc/meas/HBT/doc/html_npn13g2_VBIC/result_Spar.htm` points to `file:///home/ich/share/pdk/ihp-sg13g2/libs.doc/meas/HBT/doc/html_npn13g2_VBIC/reszoom_S22.htm` but should to `file:///home/ich/share/pdk/ihp-sg13g2/libs.doc/meas/HBT/doc/html_npn13g2_VBIC/reszoom_s22.htm`

sergeiandreyev, 2024-09-30:
Hi @olisnr, finally we can confirm the issue - could you please double check the fix on your side as well?

sergeiandreyev, 2024-10-23:
@olisnr, gentle reminder

olisnr, 2024-10-23:
Yes. its fixed. thanks!

closed 2024-10-23 by olisnr