IHP-GmbH/IHP-Open-PDK #46: VTH vs. W

claudiotalarico, 2024-02-14:
The change in VTH when sweeping W seems to be excessive (especially going from 5um to 10um).
Please see the attached plots (the device used for this analysis is the SG13 LV nmos)

[image]

[image]

[image]

The xschem's file and the matlab's file I used to generate the analysis are provided at this [link](https://github.com/claudiotalarico/EE406/tree/main/01_technology/01_vt_W/src)

bmurmann, 2024-03-20:
Below is a plot from the measured data:
[image]

It is expected that Vt is a strong function of W near the minimum allowed value (due to narrow width effects). The above data also shows this. But I would not expect is that Vt changes dramatically as the width changes from ~1-10um. The Vt value usually stabilizes for large W as shown here: https://www.semanticscholar.org/paper/Implications-of-Small-Geometry-Effects-on-%24g_m-I_D%24-Ou-Ferreira/7982fbeb3a43c594a244923438360aa5ebd1b478/figure/4

Looking at the measured data, I see identical Vt for W=0.6 and W=2um, which makes sense to me (but the current models do not reflect this). Perhaps the measured data (with potential outliers) was overfitted to generate a non-physical trend in the spice models?

dwarning, 2024-03-25:
I grubbed out few old ngspice scripts to extract Vth over W and L and rebuild them with the help of Holger.
To extract the threshold voltage a simple constant current criteria (W/L*1e-6) was used and follows in deviations compared with the IHP documented behaviour. But it confirms clear the discussed characteristic and that the models reflect the measured and documented values.
lv_nmos Vth over W with L as parameter 0.13...0.16um in linear range:
[image]

For completeness also lv_nmos Vth over L with W as parameter 2...8um in linear range:
[image]

KrzysztofHerman, 2024-03-26:
@dwarning could you please share your testcase? I would like to get the exact difference between model and the measurements.

dwarning, 2024-03-26:
No problem, but as said there are some margin in accuracy of the vth determination.

[Vth_W.txt](https://github.com/IHP-GmbH/IHP-Open-PDK/files/14755597/Vth_W.txt)
[Vth_L.txt](https://github.com/IHP-GmbH/IHP-Open-PDK/files/14755598/Vth_L.txt)

Good luck

KrzysztofHerman, 2024-07-15:
@claudiotalarico first of all we are sorry to let you wait such a long time for response. Reviewing your simulation you use W in a range up to 50u. Our models are valid up to 10u and to obtain W=50u one should use multi-finger approach. Could you please regenerate the graphics to be aligned with the one we have shared in this thread ? I mean semilogx axis for width and vertical grid of 50 mV.

hpretl, 2024-07-17:
Maybe this issue is related to https://github.com/IHP-GmbH/IHP-Open-PDK/issues/164? Could be if the number of fingers is also changed in the investigations above.