IHP-GmbH/IHP-Open-PDK #52: Layout grid

bmurmann, 2024-02-28:
In SG13G2_os_layout_rules.pdf the layout grid is specified as 5 nm, while in sg13g2.lyt it is set to 1 nm. Is this an intentional scale factor or is there some other reason for this difference?

KrzysztofHerman, 2024-03-14:
@bmurmann thank you for the comment. After some other issues with XOR cross checking of our PyCells against cells generated using commercial tools we also discovered some inconsistencies in the behavior of the tools. We will probably bring back the value of 1nm.

sergeiandreyev, 2024-03-15:
the DBU and grid are different in their purpose and setting, both in commercial PDK and in KLayout, so we decided to rollback the changes (the dbu parameter in KLayout will be changed back to 1nm)
regarding the [manufacturing] grid, we still have to understand what is the best way to show/force the users to use 5nm grid when drawing layouts..
and of course, there are several offGrid checks that we have already as part of DRC rule deck - these should find violations and actually enforce the usage of correct grid setting

sergeiandreyev, 2024-06-20:
@bmurmann, in one of the latest KLayout updates the following feature was added:
[image]
So now our KLayout tech `.lyt` file includes technology-specific default grid settings w/ the min one *strongly* set to 5nm. -> https://github.com/IHP-GmbH/IHP-Open-PDK/commit/85b2f7a030c1f1d442a1b5ec124f7cdd57020c0c
This way the grid is set by default to 5nm when you're starting KLayout in clean environment (no `klayoutrc` file present). The user still can set the grid to incorrect value.
Could you please test and provide feedback on the resolution of this issue?

bmurmann, 2024-06-22:
This works. Thank you.

closed 2024-06-22 by bmurmann