IHP-GmbH/IHP-Open-PDK #54: KLayout: PMOS and NMOS have different GatPoly Layers

vkrahul77, 2024-03-07:
In the "IHP Pycells", For Gate The PMOS has "GatPloy.lbl" while NMOS has "Gatpoly.drw" as shown:
[image]

Also, I've noticed a discrepancy in dimension input between NMOS and PMOS transistors within the tool. Specifically, while "1u" notation is accepted for PMOS, it's not permitted for NMOS. Instead, we're required to use scientific notation such as "1e-6".

ihpopenpdk, 2024-03-07:
it seems to be an issue in both `nmos` and `pmos` cells - there should be no any '.lbl' or '.bnd' `GatPoly` layers, only '.drw' and '.pin'..
Thank you for finding!

vkrahul77, 2024-03-12:
Along with the posted bug, I noticed that there are a few DRC errors in NMOS-Cell. 
Here is a list:
Min and Max size of contact should be 0.16  --> Cell has 0.14
Min Contact space should be 0.18 --> Cell has 0.16
Min Metal1 space or notch should be 0.18 --> Cell has 0.13
[image]

sergeiandreyev, 2024-05-18:
@vkrahul77 , I believe all these issues are fixed in the latest OpenPDK `dev` branch:
[image]

vkrahul77, 2024-07-10:
> Hi @vkrahul77
> <https://urldefense.com/v3/__https://github.com/vkrahul77__;!!PvDODwlR4mBZyAb0!T7BBeprlZ9Py4XdIirmXmZeCW12xNZPS-JJXRv3eEZ7R-F-Pilef9yZS5MwyJQ5OqHtLMS3wlMpXhXCdhyHZavP7$>,
> can we close the ticket? Do you have any additional comments?

Sure! Please go ahead and close the ticket.
Sincerely,
Vikas

closed 2024-07-10 by sergeiandreyev