IHP-GmbH/IHP-Open-PDK #61: Discrepancy in pin order of stdcell spice models vs. xschem symbols

bmurmann, 2024-03-27:
I am not sure of this was brought up here already. The spice pin order between these two stdcell representations do not match up:

libs.tech/xschem/sg13g2_stdcells
libs.ref/sg13g2_stdcell/spice/sg13g2_stdcell.spice

The spice file pins are in alphabetical order, but the xschem pins are not.

The following sed applied to the xschem symbols fixes the problem (courtesy Mitch Bailey):
sed -i .bak \
-e 's/@VGND @VNB @VPB @VPWR/@VDD @VSS/' \
-e 's/VGND=VGND VNB=VNB VPB=VPB VPWR=VPWR/VDD=VDD VSS=VSS/' \
-e 's/VGND VNB VPB VPWR/VDD VSS/' \
-e 's/@VDD @VSS @@Q @@Q_N/@@Q @@Q_N @VDD @VSS/' \
-e 's/\(@@RESET.*\)@@Q @@Q_N /@@Q @@Q_N \1/' *.sym

KrzysztofHerman, 2024-04-03:
Dear @bmurmann thank you for notification and the `sed` code snippet. Please checkout the recent PR #63 where the standard cells symbols were updated.

bmurmann, 2024-07-10:
Resolved.

closed 2024-07-10 by sergeiandreyev