IHP-GmbH/IHP-Open-PDK #64: Glitch in capacitance evaluation

dwarning, 2024-04-04:
The attached ngspice script [cg_glitch.txt](https://github.com/IHP-GmbH/IHP-Open-PDK/files/14862148/cg_glitch.txt)  shows Cgb, Cg_ds and Cg_dsb capacitances over Vg of a 400/0.5 lv_nmos transistor.
The strong glitch around -0.9V is in my opinion fatal and follows in time step problems for transient simulation.

The observed behaviour is for psp103 version 6.0 and actual version 8.2 [va-models](https://github.com/dwarning/VA-Models/tree/main/code/psp103/vacode) and can also be seen for the hv-nmos device.

Would be fine to see same curves from same va-code with other simulators.

[image]

Additional open question is the y-axis scale of the documented curves in [report_nmos_lv](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.doc/meas/MOS/doc/report_nmos_lv.pdf) page 401. Are this 400 pF? Because e-12 is shown.

sergeiandreyev, 2024-04-10:
Hi @dwarning,
we still have to reproduce the testcase in Xyce, as well as run/recheck this for commercial (internal) PDK models
on your second question - yes, these are 400pF

dwarning, 2024-04-10:
I see no problem with adms/Xyce:

[image]

KrzysztofHerman, 2024-04-12:
Hello, @dwarning 
could you please share the testcase for Xyce as well, there is an inconsistency with the values on Y-axis.
As for the plots of `Cgdsb` I attach simulation results where `ng` parameter is used 

[image]

the orange curve corresponds to the device with `ng=1`
`X1 _net2  _net0  _net2  _net2 sg13_lv_nmos w=400u l=0.5u`
and the blue one corresponds to the device with `ng=40` what limits `W` to `10u`
`X1 _net2  _net0  _net2  _net2 sg13_lv_nmos w=400u l=0.5u ng=40`

dwarning, 2024-04-12:
OK, scripts attached. ( spend not the time to adapt Xyce library syntax )
[cg_glitch.plt.txt](https://github.com/IHP-GmbH/IHP-Open-PDK/files/14961403/cg_glitch.plt.txt)
[cg_glitch.cir.txt](https://github.com/IHP-GmbH/IHP-Open-PDK/files/14961404/cg_glitch.cir.txt)
[cg_glitch.sp.txt](https://github.com/IHP-GmbH/IHP-Open-PDK/files/14961405/cg_glitch.sp.txt)

Even I assume that original problem has nothing to do with geometry control I used your recommended size:
`X1 _net2  _net0  _net2  _net2 sg13_lv_nmos w=400u l=0.5u ng=40`

adms - Xyce:
[image]

openvaf - ngspice:
[image]

Because the glitch is at VFB I am sure there is a bug in the openvaf compilation. Not easy to find, but possible.

sergeiandreyev, 2024-04-15:
@metroid120, could you please take a look?..

metroid120, 2024-04-15:
What I can already say, is that the bug is related to NQS effects in "PSP103_ChargesNQS.include". 
It could indeed be an OpenVAF bug, but I did not yet find the exact cause.

metroid120, 2024-04-15:
I agree that this is a serious bug. I was not able to track the issue down exactnly, but I found that 
the issue is related to the PSP current calculation in "PSP103_SPCalculation.incldue"

```
// Core PSP current calculation
    Vdsat_lim    =  3.912023005 * phit1;
    if (xg <= 0.0) begin
        qis          =  0.0;
        xgm          =  xg - x_s;
        Voxm         =  xgm * phit1;
        qeff1        =  Voxm;
        Vdsat        =  Vdsat_lim;
        Vdse         =  Vds;
    end else begin // (xg > 0)
        ...
```
For the xg>0 branch, the variabel "Vxm" should be continued smoothly, but is not. 
I do not know exactly how those model functions should work together, so it is difficult to find the root cause.
Maybe somebody else has more insight...

sergeiandreyev, 2024-04-18:
It's good that this will be tracked on OpenVAF repo, but still I propose to have this issue open here until the fix in OpenVAF will be made available.

metroid120, 2024-04-20:
Thanks to the bad weather I spend some time on this. **Folks, can you please confirm.** 

If one replaces the following three code lines in the Verilog-A

```
            SP_S_delta0 = exp(SP_S_x0); \
            SP_S_delta1 = 1.0 / SP_S_delta0; \
```
(occurs 2 times in the code)

```
            delta_1s     =  exp(x_s); \
            Es           =  1.0 / delta_1s; \
```
with 
```
            SP_S_delta0 = exp(SP_S_x0); \
            SP_S_delta1 = exp(-SP_S_x0); \
```
```
            delta_1s     =  exp(x_s); \
            Es           =  exp(-x_s); \
```
(so in total 3 occurrences), the problem is gone.

dwarning, 2024-04-20:
I can agree - this 3 modifications let the glitch disappear. Seems openvaf makes a kind of over-optimization.

metroid120, 2024-04-21:
Ok @sergeiandreyev , then I guess we have a fix/workaround. Replace the blocks of code as described above and the Glitch is gone.

I think OpenVAF generates the derivative of $y2$ in the following structure 
$y1=exp(a)$
$y2=1/exp(a)$

as

$\frac{dy2}{dx}=-(exp(a))^{-2}\frac{da}{dx}$

which causes numerical problems in contrast to the mathematically equivalent

$\frac{dy2}{dx}=-exp(-a)\frac{da}{dx}$

but this is just my guess.

Anyhow I doubt that the fundamental OpenVAF issue will be fixed soon, since the project is not active.

sergeiandreyev, 2024-04-22:
@metroid120 , thank you for taking the time to thoroughly investigate this issue over the weekend and finding a solution!

closed 2024-04-23 by sergeiandreyev

dwarning, 2024-04-23:
A note that the modified PSP model from this distribution is to use would be preferable.