IHP-GmbH/IHP-Open-PDK #80: Unsupported expression for cell sg13g2_sdfbbp_1

dnltz, 2024-04-24:
I found the following warning in OpenROAD and it looks like the stdcell lib has a type included: https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dceb7e6bd1a877182c3ba32c2e238be08368fa3f/ihp-sg13g2/libs.ref/sg13g2_stdcell/lib/sg13g2_stdcell_typ_1p50V_25C.lib#L27289

`Warning: Found unsupported expression 'SCE*SCD+SCE'*D' in pin attribute of cell 'sg13g2_sdfbbp_1' - skipping.`

I can send a fix but can someone confirm this is a typo?

dnltz, 2024-04-24:
@KrzysztofHerman

KrzysztofHerman, 2024-04-25:
@dnltz thank you for reproducing this as well because we are aware of this warning. The person in charge has it in his pipeline but it you can provide the fix such as typo go ahead and do so so we can check it out using other tools as well.

dnltz, 2024-04-26:
See https://github.com/IHP-GmbH/IHP-Open-PDK/pull/83

sergeiandreyev, 2024-04-29:
Hi @dnltz , this is not a typo - this is an inversion, if you remove this symbol the function will not be correct. An option will be to change this symbol to `!`: `(SCE*SCD)+(!SCE*D)`
Could you please try to change the function:
```
ff (IQ,IQN) {
      clear : "!RESET_B";
      clear_preset_var1 : "H";
      clear_preset_var2 : "L";
      clocked_on : "CLK";
      next_state : "(SCE*SCD)+(!SCE*D)";
      preset : "!SET_B";
      }
 ```
 and give us your feedback whether it passes now.

dnltz, 2024-05-03:
@sergeiandreyev still the same warning. I wonder if this expression is allowed at all and should be replaced by boolean operations since it defines the next state of a flip-flop.

sergeiandreyev, 2024-05-05:
hm, ok.. the `*` and `+` boolean operators are valid/allowed if you use commercial tools, we don't see any issues internally.
then how about this one? `"(SCE&SCD)|(!SCE&D)"` -> changed `*` to `&` and `+` to `|`
could you please check if the above syntax will work on your side?
btw, this is how sky130 scan flops are described:
```
ff ("IQ","IQ_N") {
            clear : "!RESET_B";
            clear_preset_var1 : "H";
            clear_preset_var2 : "L";
            clocked_on : "!CLK_N";
            next_state : "(D&!SCE) | (SCD&SCE)";
            preset : "!SET_B";
        }
```

dnltz, 2024-05-06:
@sergeiandreyev this statement  is also illegal. I figured out only one pin is OK but as soon as you add another one it's bad.

sergeiandreyev, 2024-05-06:
Hm, then I don't get how the SkyWater sky130 scan flops are used in OpenROAD.. 
Does it support scan chains?

sergeiandreyev, 2024-05-06:
next_state w/ one pin is for regular flop
when you have a scan flop, scan signals should be defined in the parameter

noherbrferurtth, 2024-05-06:
> This is an external E-Mail. Do not click any links or open any attachments unless you trust the sender and know the content is safe. When you are in doubt please contact the IT-Team by using TTS or forward this E-Mail to ***@***.******@***.***> .

So far there is no implementation of a scan chain with Open Source tools. To my knowledge.

Dr.-Ing. Norbert Herfurth
Scientist
Technology / High-Performance Si-Technologies
IHP

Fon: +49 (335) 5625-525
Fax: +49 (335) 5625-327
E-Mail: ***@***.******@***.***>
Website: www.ihp-microelectronics.com<https://www.ihp-microelectronics.com>

[cid:SocialLink_Facebook_32x32_f848f5b0-52e0-4dda-a869-9f2278b71813.png]<https://www.facebook.com/IHPFFO> [cid:SocialLink_Twitter_32x32_242a0cd4-6fff-468a-8361-f726eb7cd314.png] <https://twitter.com/waferffo>  [cid:SocialLink_Linkedin_32x32_33a16910-12a4-40c5-ba5d-13f0b7ac8b86.png] <https://www.linkedin.com/company/ihp>

IHP GmbH
Leibniz Institute for High Performance Microelectronics
Leibniz-Institut für innovative Mikroelektronik
Im Technologiepark 25
15236 Frankfurt (Oder)
Germany

Geschäftsführer: Prof. Dr.-Ing. Gerhard Kahmen, Nicolas Hübener, Vorsitzende des Aufsichtsrates: Dr. Inge Schlotzhauer
Handelsregister: HRB 1982 Amtsgericht Frankfurt (Oder) , Steuernummer: 061/125/00152, Umsatzsteuer-ID: DE 138996546

Die IHP GmbH ist Mitglied der Leibniz-Gemeinschaft und wird von der Bundesrepublik Deutschland und dem Land Brandenburg institutionell finanziert.

Diese E-Mail enthält vertrauliche und / oder rechtlich geschützte Informationen. Wenn Sie nicht der richtige Adressat sind oder diese E-Mail irrtümlich erhalten haben, informieren Sie bitte sofort den Absender und vernichten Sie diese Mail. Das unerlaubte Kopieren sowie die unbefugte Weitergabe dieser Mail sind nicht gestattet.

This e-mail contains confidential and / or privileged information. If you are not the intended recipient or have received this e-mail in error, please notify the sender immediately and destroy this e-mail. Any unauthorized copying, disclosure or distribution of the material in this e-mail is strictly forbidden.

mmlouerat, 2024-05-07:
Goodday,

I am writing to you on behalf of Pr Naohiko Shimizu (naohiko.shimizu@lip6.fr) who is currently for one year at Sorbonne Université-LIP6, within the Open Source EDA Coriolis Team.

Pr Shimizu used a scan path insertion tool available in the Alliance tool chain.
It is called "scapin", not so young today.

Alliance sources may be found:
https://github.com/lip6/alliance

here is the man page of scapin:

SCAPIN(1)                                     CAO-VLSI Reference Manual                                     SCAPIN(1)

NAME
       scapin - Scan path insertion

ORIGIN
       This  software belongs to the  ALLIANCE CAD SYSTEM developed by the ASIM team at LIP6 laboratory of Université
       Pierre et Marie CURIE, in Paris, France.

       Web : http://asim.lip6.fr/recherche/alliance/
       E-mail : [alliance-users@asim.lip6.fr](mailto:alliance-users@asim.lip6.fr)

SYNOPSIS
       scapin [-VRB] [-P file] Input_name Path_name Output_name

DESCRIPTION
       SCAPIN is an automatic scan path generator for gate level netlists.  SCAPIN inserts a scan path in the netlist
       Input_name  and drives a new netlist Output_name.  This scan path contains all registers specified in the file
       Path_name.path (see below for the exact syntax).  SCAPIN adds also 3 new connectors in the  netlist:  scan_in,
       scan_out  and scan_test in order to control the scan path.  Eventually SCAPIN adds an output buffer before the
       output connector scan_out (option -B).

mmlouerat, 2024-05-14:
Hi,
There are also newer test tools that will be presented at FSIC2024:
- https://github.com/AUCOHL/Fault for scan path insertion
- https://github.com/Coloquinte/quaigh ATPG, should you need it.

noherbrferurtth, 2024-05-14:
thanks a lot for the link! I was not aware. This is very helpful for the upcoming HSM we are planing.

sergeiandreyev, 2024-06-22:
> So far there is no implementation of a scan chain with Open Source tools. To my knowledge.

apart from the `scapin` tool that @mmlouerat mentioned there is an update after FSiC'24: https://github.com/AUCOHL/Fault