IHP-GmbH/IHP-Open-PDK #86: Implement method of splitting substrate into local islands (using `DigiSub`)

hpretl, 2024-04-28:
In a typical IC design, different VSS nets are used in different areas for blocks requiring isolation (e.g., digital blocks vs. analog blocks), leading to p-taps shorting these VSS nets via the (common) substrate node if no measures are taken. 

A common method is to split the substrate into local VSS islands using a marking layer to circumvent these problems.

In SG13G2 this marking layer is called `DigiSub`. Please support this layer in LVS and PEX.

sergeiandreyev, 2024-05-17:
Hello @hpretl , could you please confirm that with the latest LVS update this enhancement is implemented?

hpretl, 2024-05-19:
Hi @sergeiandreyev I can check, but this will take quite a while.

KrzysztofHerman, 2024-12-18:
@hpretl a gentle reminder on this action