IHP-GmbH/IHP-Open-PDK #88: dantenna & dpantenna netlisting

bmurmann, 2024-04-28:
The pin order of these diodes in spice netlists is alphabetical, and currently gives the pin order cathode, anode. This is incompatible with standard spice, where diodes are instantiated as anode, cathode.

This problem has trickled down into the LVS, which now expects the wrong cathode, anode order. The existing netlist for the IO cells uses the proper spice ordering and therefore does pass LVS.

I strongly recommend going with the spice standard; anything else will be extremely confusing.

FaragElsayed2, 2024-05-02:
Done in #93 

Cc @atorkmabrains

sergeiandreyev, 2024-05-17:
@bmurmann, could you please confirm that with the latest LVS update the issue is fixed?

bmurmann, 2024-05-17:
Confirmed, this is fixed.

closed 2024-05-18 by sergeiandreyev