IHP-GmbH/IHP-Open-PDK #90: RPPD not recognized on n-well (alpha LVS)

bmurmann, 2024-04-29:
Title should be self-explanatory. The derivation file for RPPD excludes n-well as a valid region for RPPD (which I don't think is correct).

FaragElsayed2, 2024-05-02:
Hi Dr. @bmurmann 

Is this applicable only to `rppd`, or does it apply to all poly resistors `(rhigh, rppd, rsil)`?

bmurmann, 2024-05-02:
Yes, I believe all poly resistors should be allowed on n-well, but you may want to double check with @KrzysztofHerman.

KrzysztofHerman, 2024-05-02:
@FaragElsayed2 @bmurmann I confirm, you are free to use all resistor in both regions: `p-sub` and `n-well`.

sergeiandreyev, 2024-05-07:
Hi all, as I understand, this should be fixed in the latest update of LVS deck..
@bmurmann, could you please confirm?

bmurmann, 2024-05-07:
@sergeiandreyev Yes, confirmed.