IHP-GmbH/IHP-Open-PDK #94: Stdcell pin order

bmurmann, 2024-04-30:
sg13g2_stdcell.cdl has the wrong supply pin order for inv_4 and inv_8. The order should be VDD, VSS.

************************************************************************
* Library Name: sg13g2_stdcell
* Cell Name:    sg13g2_inv_2
* View Name:    schematic
************************************************************************

.SUBCKT sg13g2_inv_2 A VDD VSS Y
*.PININFO A:I Y:O VDD:B VSS:B
MX1 Y A VSS VSS sg13_lv_nmos m=2 w=740.00n l=130.00n ng=1
MX0 Y A VDD VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
.ENDS

************************************************************************
* Library Name: sg13g2_stdcell
* Cell Name:    sg13g2_inv_4
* View Name:    schematic
************************************************************************

.SUBCKT sg13g2_inv_4 A VSS VDD Y
*.PININFO A:I Y:O VDD:B VSS:B
MP0 Y A VDD VDD sg13_lv_pmos m=4 w=1.12u l=130.00n ng=1
MN0 Y A VSS VSS sg13_lv_nmos m=4 w=740.00n l=130.00n ng=1
.ENDS

************************************************************************
* Library Name: sg13g2_stdcell
* Cell Name:    sg13g2_inv_8
* View Name:    schematic
************************************************************************

.SUBCKT sg13g2_inv_8 A VSS VDD Y
*.PININFO A:I Y:O VDD:B VSS:B
MX1 Y A VSS VSS sg13_lv_nmos m=8 w=740.00n l=130.00n ng=1
MX0 Y A VDD VDD sg13_lv_pmos m=8 w=1.12u l=130.00n ng=1
.ENDS

sergeiandreyev, 2024-05-03:
Hello @bmurmann , this should be fixed now (https://github.com/IHP-GmbH/IHP-Open-PDK/commit/5f02042b601a174e58aefb474a3a5760ab3f7eb6)

sergeiandreyev, 2024-05-08:
@bmurmann, could you please confirm that it's fixed - then I'll close the issue

bmurmann, 2024-05-08:
Confirmed, it's fixed.

closed 2024-05-09 by sergeiandreyev