IHP-GmbH/IHP-Open-PDK #95: Use of wire_load_table prevent import to ORFS

maliberty, 2024-05-01:
I tried to update ORFS's copy of the IHP pdk to resolve #92 .  However sg13g2_stdcell_typ_1p20V_25C.lib is using wire_load_table which abc does not support and gives:

```
Error: Cannot find wire load model "1k".
```

abc does support wire_load which is what I see in the ORFS. @KrzysztofHerman did you change this when importing to ORFS?

mi-and-core, 2024-05-01:
@maliberty When we discussed this in slack, it was changed in OR to wire_load(), but the IHP libraries were not changed. Yesterday IHP's dev branch was updated regarding this. Later on we will see the change in their main branch

maliberty, 2024-05-01:
Thanks, sorry I forgot.  I update ORFS's copy while preserving the WL change.  It sounds like it will be easier to update in the future.

closed 2024-05-01 by maliberty