IHP-GmbH/IHP-Open-PDK #100: RPPD extraction (alpha LVS)

bmurmann, 2024-05-02:
A series combination of RPPD (as shown below) matches the schematic only with option --no_simplify. But this option won't work for many other parts of a real circuit. Any remedies that can be implemented?

My test cases has 26 resistors in series in the layout and 26 resistors in series in the schematic. When any simplifications are turned on, there is no more match, since the derived simplifications between sch and lay differ.

[image]

sergeiandreyev, 2024-05-17:
Hello @bmurmann, could you please confirm that with the latest LVS update the issue is fixed?

bmurmann, 2024-05-17:
Confirmed, this is fixed.

closed 2024-05-18 by sergeiandreyev