IHP-GmbH/IHP-Open-PDK #96: Cap extraction (alpha LVS)

bmurmann, 2024-05-02:
It looks like mimcaps don't get recognized when there is a piece of poly underneath. I am am not sure I understand this long list of exclusions for a component high up in the stack. I think this is a bug (but it needs confirmation).

cap_exc = nsd_drw.join(heattrans_drw).join(trans_drw)
           .join(emwind_drw).join(emwihv_drw).join(heatres_drw)
            .join(salblock_drw).join(polyres_drw).join(extblock_drw)
            .join(res_drw).join(activ_mask).join(recog_diode)
            .join(recog_esd).join(ind_drw).join(ind_pin)
            .join(substrate_drw)

FaragElsayed2, 2024-05-02:
Hi Dr. @bmurmann 

I have constructed the following truth table based on my knowledge and after discussion with @atorkmabrains.

@KrzysztofHerman Could you please confirm if we could use the `mimcap & inductor` on the top of other devices? 
[image]
[image]
[image]

atorkmabrains, 2024-05-02:
Thanks Dr. @bmurmann for the feedback.

The recommendation that I followed in this behavior, is the avoidance of adding any device underneath MIM Caps to make sure that we don't create unintended behavior from the RF design, signal integrity, etc... perspective. 

BTW, Inductors in this technology has `pwell_block` to push the self-resonance frequency higher for example. I understand such choice should be made on the design side rather than on the tool side.

Happy to adopt removing such limitations. But I believe we need IHP team confirmation.

@KrzysztofHerman Could you please confirm which behvior you need?

bmurmann, 2024-05-02:
As you say, I think these are really design decisions. I have put transistors underneath MIMcaps in many projects (and this is one of the nice features offered by MIMcaps for design destiny). If there is no reason from the technology/manufacturing side, the LVS should allow it.

atorkmabrains, 2024-05-02:
@KrzysztofHerman Could you please confirm from IHP side? Otherwise, we will take @bmurmann advise.

KrzysztofHerman, 2024-05-02:
@bmurmann and @atorkmabrains thank you for taking care about this details. In general I would agree with @bmurmann having similar experiences however I do not have any in RF so please let me consult our RF designers and PDK developers to have some more information.

atorkmabrains, 2024-05-02:
@KrzysztofHerman I think the question here: Is there any technological concern from IHP side about allowing designers to add devices under MIM?

atorkmabrains, 2024-05-02:
@KrzysztofHerman Could you please reopen and keep it until we update the LVS rule deck?

hpretl, 2024-05-02:
@KrzysztofHerman RF designer here :-): For MIMCAP full flexibility should be allowed, i.e. can be placed above anything. I often place it above MOSFET in switched-capacitor applications to minimize routing parasitics (just need to go straight down to switch).

sergeiandreyev, 2024-05-17:
@bmurmann, could you please confirm that with the latest LVS update the issue is fixed?

bmurmann, 2024-05-17:
Confirmed, this is fixed.

closed 2024-05-18 by sergeiandreyev