IHP-GmbH/IHP-Open-PDK #99: Slit rules in process spec

hpretl, 2024-05-02:
Please add clarifying language whether the slitting rules `Slt.e` and `Slt.e1` in https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.doc/doc/SG13G2_os_layout_rules.pdf are true for all metals, or just `TopMetal2`.

In addition, maybe the layer names should be mentioned which create these exceptions.

stefansimon42, 2024-05-06:
Thanks @hpretl for the question.

You are absolutely right. A more precise description of the affected levels can be very helpful.

Slt.e affects all metals that are connected to the pad. However, I'm not entirely sure whether this can be easily implemented with the standard DRC tools. This needs to be evaluated separately.

Slt.e1 here only affects the metals that a MIM capacitor is actually composed of, i.e. Metal5 and TopMetal1.

hpretl, 2024-05-06:
@stefansimon42 Do you mean "Slt.e affects all metals covered by `pad`"? If so, then this could be checked by `AND`ing the respective layers. Or is it really all metals connected to pad, but this would then progress far into the chip.

stefansimon42, 2024-05-07:
`AND`ing the pad layer and all metal layers may select too many polygons. 

For example, a pad can extend from layer Metal3 to TopMetal2. These layers should not be slotted. However, if a ground plane runs underneath in Metal1, this must be slotted. It is thus important to also check the connectivity between Pad-TopMetal2 and the underlying metal layers and then define which layers are to be slotted.

hpretl, 2024-05-07:
@stefansimon42 Thanks, yes, this totally makes sense. So the rule is (formulated a bit clumsy) `[(pad AND TM2) connected layers] AND pad]`.

KrzysztofHerman, 2024-12-18:
@stefansimon42 will we provide some updates in our documentation regarding this issue?

stefansimon42, 2024-12-19:
We are planning a complete revision of the general pad rules in 2025, which will take into account the issues discussed above.