IHP-GmbH/IHP-Open-PDK #101: sg13g2_IOPadVdd

bmurmann, 2024-05-03:
I have managed to get many of the IOPads in the dev branch through the KLayout LVS, but it seems that the IOPadVdd cell has some issues such as some dangling drain/source nodes and missing devices (relative to .spi description). Can someone confirm whether this cell is the latest version and passes LVS on your side (commercial or open source)?

* This is the circuit description. I don't see the DCN & DCP devices in the layout.
* sg13g2_IOPadVdd
.subckt sg13g2_IOPadVdd vss vdd iovss iovdd
Xnclamp iovss iovdd vdd ngate sg13g2_Clamp_N43N43D4R
Xrcres vdd res_cap sg13g2_RCClampResistor
Xrcinv vdd iovss res_cap ngate sg13g2_RCClampInverter
Xdcndiode iovss vdd iovdd sg13g2_DCNDiode
Xdcpdiode vdd iovdd iovss sg13g2_DCPDiode
.ends sg13g2_IOPadVdd

stafverhaegen-chipflow, 2024-05-03:
I can confirm that the netlist is wrong. The DCN & DCP should not be in the netlist. Does it match if you remove them ?

The KLayout LVS was not available during the design if the IO cells. So un-conventional verification was done. It was semi-manual and actually checked that the extracted netlist with Calibre delivered the right functionality. As the diodes don't impact functionality this discrepancy was not seen.

The IOPadVdd cell is the cell was the latest in dev before previous tape-out. Will do some more checking next week and then submit a patch.

stafverhaegen-chipflow, 2024-05-03:
> I have managed to get many of the IOPads in the dev branch through the KLayout LVS

Actually that is good to know. I thought that the handling of the ptaps in the Calibre deck would be not so easy to port to KLayout.

bmurmann, 2024-05-03:
Thank you for confirming. Below is a summary of what I had to do pass LVS for this cell:

Change cdl as follows:
```
.subckt sg13g2_IOPadVdd vss vdd iovss iovdd
Xnclamp iovss iovdd vdd ngate sg13g2_Clamp_N43N43D4R
Xrcres vdd res_cap sg13g2_RCClampResistor
*** Inverter is connected to iovdd in layout (does not seem to be correct)
*Xrcinv vdd iovss res_cap ngate sg13g2_RCClampInverter
Xrcinv iovdd iovss res_cap ngate sg13g2_RCClampInverter
*** These are not in the layout
*Xdcndiode iovss vdd iovdd sg13g2_DCNDiode
*Xdcpdiode vdd iovdd iovss sg13g2_DCPDiode
.ends sg13g2_IOPadVdd
```
Add missing M2 connections to the pad in the layout (showing just skinny paths for illustration):
[image]

bmurmann, 2024-05-03:
Here are also some notes on what I had to do to pass the KLayout LVS for other IO Cells (work in progress):
https://github.com/bmurmann/EE628/blob/a659c76b3c41ed46662405ceca7dbc085c4eda4c/5_Design/4_Layout/sg13g2_io_bm_hacks.md

stafverhaegen-chipflow, 2024-05-06:
> *** Inverter is connected to iovdd in layout (does not seem to be correct)
> *Xrcinv vdd iovss res_cap ngate sg13g2_RCClampInverter
> Xrcinv iovdd iovss res_cap ngate sg13g2_RCClampInverter

Mmm, this is no good. Seems I did not use right net list when verifying this circuit.
This means that both the NMOS and the PMOS of this inverter will be turned on, resulting in a DC current from IOVDD to IOVSS.
Kind of lucky that the supposed connection to VDD on the clamp was accidentally left floating. Otherwise this bodged inverter could have caused much more havoc as the drive strength of this clamp is much higher than the inverters one.

I should be able to fix the cell quite quickly; I hope to do that today.

stafverhaegen-chipflow, 2024-05-06:
* [sg13g2_io.gds.gz](https://github.com/IHP-GmbH/IHP-Open-PDK/files/15219786/sg13g2_io.gds.gz)
* [sg13g2_io.spi.gz](https://github.com/IHP-GmbH/IHP-Open-PDK/files/15219828/sg13g2_io.spi.gz)

These files should fix the problems for `IOPadVdd`; you'll need to apply the same magic with your added layers as for the other cells though. Fixing the latter will need a little more discussion.
Also I seem to have messed up DRC check overhere so I need to handle that first before doing an MR to the `IHP-Open-PDK` dev branch with the fixed gds.

bmurmann, 2024-05-07:
Thank you, Staf. This fix worked fine.

stafverhaegen-chipflow, 2024-05-13:
@bmurmann 
Only see your updated comment now as one does not get notification if a comment is changed.  
I did #109 but that is very similar to the gds I uploaded here just the vdd connection is segmented.
My check with Calibre does seem to indicate bulk is vdd for the transistor but I will do a cross-check again.
How can this problem only be seen when integrating with the other pads ?

bmurmann, 2024-05-13:
@stafverhaegen-chipflow The integration with other cells connects all the IOVDDs together and gives that net an LVS port label in my setup. That made the cell fail LVS since the netlist has the bulk at VDD. Not exactly sure why it passed standalone, but this was also with the alpha lvs and perhaps I overlooked something. Best to ignore this as the desired connectivity is clear. I had to add a new wire in the cell to drag VDD to that bulk location; it was not natively present near that well, if I remember correctly.

bmurmann, 2024-05-13:
Here's what I did to make the VDD connection to the well, note the skinny vertical M2 going up to the VDD rail on M3. I guess I could have come down from Topmetal2 as well.

[image]

stafverhaegen-chipflow, 2024-05-14:
There should already be a connection on metal2 there. This is latest layout from #109 with the connection segmented:
[image]

stafverhaegen-chipflow, 2024-05-14:
And in the file uploaded here it was not segmented and gave DRC error:
[image]

bmurmann, 2024-05-14:
OK. We may want to close this thread since what I wrote is about the first modified version before you did the latest updates. Best to have this verified by others, since I have not looked at the latest.

FatsieFS, 2024-05-14:
Thank you very much anyway. I would be aware of the problem without the hard work you put into this.

KrzysztofHerman, 2024-05-21:
Closing the issue. Great work!

closed 2024-05-21 by KrzysztofHerman