IHP-GmbH/IHP-Open-PDK #108: SG13G2 Standard Cells

FaragElsayed2, 2024-05-12:
## SG13G2 Standard Cells

It seems that there might be connectivity issues with the cdl files in the cells below.

### 1.sg13g2_nand2b_2

**Layout**
[image]

**Current Cdl**
```
.SUBCKT sg13g2_nand2b_2 A_N B VDD VSS Y
*.PININFO A_N:I B:I Y:O VDD:B VSS:B
MX0 Y A VDD VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
MX1 A A_N VDD VDD sg13_lv_pmos m=1 w=840.00n l=130.00n ng=1
MX3 Y B VDD VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
MX2 Y A net1 VSS sg13_lv_nmos m=2 w=720.00n l=130.00n ng=1
MX4 A A_N VSS VSS sg13_lv_nmos m=1 w=550.00n l=130.00n ng=1
MX5 net1 B VSS VSS sg13_lv_nmos m=2 w=720.00n l=130.00n ng=1
.ENDS
```

**Expected Cdl**

```
.SUBCKT sg13g2_nand2b_2 A_N B VDD VSS Y
*.PININFO A_N:I B:I Y:O VDD:B VSS:B
MX0 Y A VDD VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
MX1 A A_N VDD VDD sg13_lv_pmos m=1 w=840.00n l=130.00n ng=1
MX3 Y B VDD VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
MX2 Y B net1 VSS sg13_lv_nmos m=2 w=720.00n l=130.00n ng=1
MX4 A A_N VSS VSS sg13_lv_nmos m=1 w=550.00n l=130.00n ng=1
MX5 net1 A VSS VSS sg13_lv_nmos m=2 w=720.00n l=130.00n ng=1
.ENDS
```

**Changes**

- Flipping the gate connection between `MX2` and `MX5`.

===================================================================

### 2.sg13g2_nor2b_1

**Layout**
[image]

**Current Cdl**
```
.SUBCKT sg13g2_nor2b_1 A B_N VDD VSS Y
*.PININFO A:I B_N:I Y:O VDD:B VSS:B
MN0 B B_N VSS VSS sg13_lv_nmos m=1 w=550.00n l=130.00n ng=1
MX0 Y A VSS VSS sg13_lv_nmos m=1 w=740.00n l=130.00n ng=1
MX3 Y B VSS VSS sg13_lv_nmos m=1 w=740.00n l=130.00n ng=1
MP0 B B_N VDD VDD sg13_lv_pmos m=1 w=840.00n l=130.00n ng=1
MX1 net1 A VDD VDD sg13_lv_pmos m=1 w=1.12e-06 l=130.00n ng=1
MX2 Y B net1 VDD sg13_lv_pmos m=1 w=1.12e-06 l=130.00n ng=1
.ENDS
```

**Expected Cdl**

```
.SUBCKT sg13g2_nor2b_1 A B_N VDD VSS Y
*.PININFO A:I B_N:I Y:O VDD:B VSS:B
MN0 B B_N VSS VSS sg13_lv_nmos m=1 w=550.00n l=130.00n ng=1
MX0 Y A VSS VSS sg13_lv_nmos m=1 w=740.00n l=130.00n ng=1
MX3 Y B VSS VSS sg13_lv_nmos m=1 w=740.00n l=130.00n ng=1
MP0 B B_N VDD VDD sg13_lv_pmos m=1 w=840.00n l=130.00n ng=1
MX1 net1 B VDD VDD sg13_lv_pmos m=1 w=1.12e-06 l=130.00n ng=1
MX2 Y A net1 VDD sg13_lv_pmos m=1 w=1.12e-06 l=130.00n ng=1
.ENDS
```

**Changes**

- Flipping the gate connection between `MX1` and `MX2`.

===================================================================

### 3.sg13g2_nor2b_2

**Layout**
[image]

**Current Cdl**
```
.SUBCKT sg13g2_nor2b_2 A B_N VDD VSS Y
*.PININFO A:I B_N:I Y:O VDD:B VSS:B
MN0 B B_N VSS VSS sg13_lv_nmos m=1 w=640.00n l=130.00n ng=1
MX0 Y A VSS VSS sg13_lv_nmos m=2 w=720.00n l=130.00n ng=1
MX3 Y B VSS VSS sg13_lv_nmos m=2 w=720.00n l=130.00n ng=1
MP0 B B_N VDD VDD sg13_lv_pmos m=1 w=1.000u l=130.00n ng=1
MX1 net1 A VDD VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
MX2 Y B net1 VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
.ENDS
```

**Expected Cdl**

```
.SUBCKT sg13g2_nor2b_2 A B_N VDD VSS Y
*.PININFO A:I B_N:I Y:O VDD:B VSS:B
MN0 B B_N VSS VSS sg13_lv_nmos m=1 w=640.00n l=130.00n ng=1
MX0 Y A VSS VSS sg13_lv_nmos m=2 w=720.00n l=130.00n ng=1
MX3 Y B VSS VSS sg13_lv_nmos m=2 w=720.00n l=130.00n ng=1
MP0 B B_N VDD VDD sg13_lv_pmos m=1 w=1.000u l=130.00n ng=1
MX1 net1 B VDD VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
MX2 Y A net1 VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
.ENDS
```

**Changes**

- Flipping the gate connection between `MX1` and `MX2`.

sergeiandreyev, 2024-05-13:
Hello @FaragElsayed2, it seems that there is an issue how KLayout LVS deck is recognizing the nets in these cells. I just checked the first one `nand2b_2`, we don't see violations using commercial tools, also if looking at KLayout LVS run:
[image]
The highlighted net is internal `A` net, but here is recognized as `B` in the netlist.
On the below screenshot the net `B` is incorrectly recognized as `A`:
[image]
We can have a call if you need

atorkmabrains, 2024-05-13:
@sergeiandreyev I believe we need to review this over a meeting. LVS seems to work correctly from my end.

sergeiandreyev, 2024-05-15:
Hi @FaragElsayed2 , could you please check on your side that the issue is fixed - then I'll close this item as resolved.

FaragElsayed2, 2024-05-15:
@sergeiandreyev, yes, it's fixed now. Thanks!

closed 2024-05-15 by FaragElsayed2