IHP-GmbH/IHP-Open-PDK #112: Smaller SRAMs

SteffenReith, 2024-05-16:
To support e.g. smaller CPU-architectures as RISC-V (e.g. 16 and 8 bit systems) it would be nice to have smaller sram-macros. Nice sizes would be 4kx16, 1kx16, 4kx8 and 1kx8

noherbrferurtth, 2024-05-16:
Thank you Steffen. Your feedback is much appreciated and comes at just the right time. We are currently evaluating the possibilities of an open SRAM generator.

sergeiandreyev, 2024-07-09:
Hi @SteffenReith, what is the value for `Column Multiplex Level` that you need to have in requested SRAM blocks? What we've delivered so far inside the OpenPDK is with CML=2, however the generator is not able to process the 4K x 16 and 4K x 8 sizes with this CML value..
The only option for 4K size macros is to increase CML at least to 3

sergeiandreyev, 2024-07-31:
Hi @SteffenReith, new SRAM blocks are available in `dev` branch

sergeiandreyev, 2024-08-08:
Hi @SteffenReith, could you please check that the generated SRAMs are as expected, so we can close the issue as 'resolved'

sergeiandreyev, 2024-09-11:
Hi @SteffenReith, gentle reminder on:

> could you please check that the generated SRAMs are as expected, so we can close the issue as 'resolved'