IHP-GmbH/IHP-Open-PDK #114: For loops that do not use loop variables in contactArray function.

eskiyerli, 2024-05-17:
In this code, there are *for* loops for `i` and `j` variables, but there is nothing in the loops that refer to these variables. 
#***********************************************************************************************************************
# contactArray
#***********************************************************************************************************************
def contactArray(self, pathLayer, contLayer, xl, yl, xh, yh, ox, oy, ws, ds):
    eps = self.tech.getTechParams()['epsilon1']

    w = xh-xl
    h = yh-yl

    mlist = list()

    nx = floor((w-ox*2+ds)/(ws+ds)+eps)
    if (nx <= 0) :
        return mlist

    dsx = 0
    if (nx == 1) :
        dsx = 0
    else :
        dsx = (w-ox*2-ws*nx)/(nx-1)

    ny = floor((h-oy*2+ds)/(ws+ds)+eps)
    if (ny <= 0) :
        return mlist

    dsy = 0
    if (ny == 1) :
        dsy = 0
    else :
        dsy = (h-oy*2-ws*ny)/(ny-1)

    x = 0
    if (nx == 1) :
        x = (w-ws)/2
    else :
        x = ox

    if pathLayer :
        mlist.append(dbCreateRect(self, pathLayer, Box(xl, yl, xh, yh)))

    **for i in range(int(nx)) :**
    #for(i=1; i<=nx; i++) {
        y = 0
        if ny == 1 :
            y = (h-ws)/2
        else :
            y = oy

        **for j in range(int(ny)) :**
        #for(j=1; j<=ny; j++) {
            mlist.append(dbCreateRect(self, contLayer, Box(xl+tog(x), yl+tog(y), xl+tog(x+ws), yl+tog(y+ws))))
            y = y+ws+dsy

        x = x+ws+dsx

    if pathLayer :
        mlist.append(dbCreateRect(self, pathLayer, Box(xl, yl, xh, yh)))

    return mlist

sergeiandreyev, 2024-05-17:
@eskiyerli, thank you for the question! there is no need to use them, the `i` and `j` for-loop variables are used here just to limit the loop run to a specific number of iterations, nothing more. Still as you can see there was an attempt to tune this (`for(i=1; i<=nx; i++) {` lines are commented)

sergeiandreyev, 2024-05-21:
Hello @eskiyerli, does it answer your question?

closed 2024-05-21 by sergeiandreyev

Nado15, 2024-05-29:
The Pythonic way to handle unused indices is to use an underscore (_):
```
for _ in range(int(nx)) :     
        for _ in range(int(ny)) :   
```

sergeiandreyev, 2024-06-15:
please note that this is not a *pure* from the ground up Python code, but rather a conversion from the commercial PDK where we have Pcells in one well-known specialized programming language for EDA - you can see this from the function names ;)
And we're trying to make the style of this code as interoperable as possible, easy to translate / convert between tools