IHP-GmbH/IHP-Open-PDK #116: DRC violation in sealring

dnltz, 2024-05-17:
Layout rules define GFil.d with a gap of 1.1u between GatPoly:filler and pSD. The current  sealring only has 1.0u and therefore fails this check.

[image]

dnltz, 2024-05-17:
Only the horizontal filler cells are affected. Verticals have a gap of 1.1u.

dnltz, 2024-05-17:
https://github.com/IHP-GmbH/IHP-Open-PDK/blob/f5c16b18d8d7048e65e7bd63d392d065fec73d4b/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp/sealring_code.py#L191

sergeiandreyev, 2024-05-17:
Unfortunately, after review we see that we have a disconnect between the `sealring` in the OpenPDK and the internal one:
[image]
we're going to remove the internal filler generation code altogether from the `sealring` Pycell, this way the external script should be used and also we will get rid of these errors that you're mentioning here

dnltz, 2024-05-18:
Should be solved with https://github.com/IHP-GmbH/IHP-Open-PDK/pull/117

@sergeiandreyev can we close it?

sergeiandreyev, 2024-05-18:
yes, sure, thank you again!
in the meantime I'll cross check the updated Pycell with the internal implementation

closed 2024-05-18 by sergeiandreyev