IHP-GmbH/IHP-Open-PDK #148: Which Open Source Simulation Tool to Use?

MaxenceBouvier, 2024-06-24:
Hi,

I am trying to simulate a post-synthesis design using either [Verilator](https://www.veripool.org/verilator/) or [icarus verilog (iVerilog)](https://github.com/steveicarus/iverilog) tools.

The design has been synthesized with YOSYS simulator and in the end instantiates some sg13g2_dfrbp_1 cells.

Unfortunately, the behavioral model of these cells rely on:
 - [User Defined Primitives (UDP) description.](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/5a42d03194e8c98558f4e34538338a60550f89b9/ihp-sg13g2/libs.ref/sg13g2_stdcell/verilog/sg13g2_stdcell.v#L459)
 - [Standard Delay Format (SDF) timing definitions.](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/5a42d03194e8c98558f4e34538338a60550f89b9/ihp-sg13g2/libs.ref/sg13g2_stdcell/verilog/sg13g2_stdcell.v#L1766)
  
 So, I do not succeed running a simulation for the following reasons:
 -  is not supported by Icarus Verilog. ([here](https://github.com/steveicarus/iverilog/issues/943))
 - Verilator cannot read UDP Verilog description. ([here](https://github.com/verilator/verilator/issues/468))
 
What do you recommend as a solution, apart from redefining the behavior of sg13g2_dfrbp_1 as simple D Flip Flops?

Thanks for your help.

KrzysztofHerman, 2024-06-24:
Hi @MaxenceBouvier thank you for commenting on that. Please take a look at the  [OpenROAD-flow-scripts](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts) and our platform, namely  `ihp-sg13g2`, available there. In short the technology mapping is done by Yosys using cell definition included in `*.lib` file.

mole99, 2024-06-24:
I would like to add that SDF is partially supported by Icarus Verilog. I did some tests here with Sky130A: https://github.com/mole99/interconnect-tests
Unfortunately, no timing checks are implemented yet.

Simulating at gate level without timing information should be no problem with Icarus Verilog. I just read the file `sg13g2_stdcell.v` without errors, there are of course warnings indicating the missing timing checks.

Make sure you are using a current version of Icarus Verilog.

Blebowski, 2024-06-26:
Hi,

there is a VHDL simulator where the author started to integrate Verilog support. I have started to work on
SDF annotation last summer, but the work is stalled for now.
https://github.com/nickg/nvc/pull/757

It is not where you need it yet, but it is slowly moving there.

MaxenceBouvier, 2024-06-26:
Thank you all for yours answers!

> Hi @MaxenceBouvier thank you for commenting on that. Please take a look at the [OpenROAD-flow-scripts](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts) and our platform, namely [...]

Thanks @KrzysztofHerman, indeed, I am able to synthesize and techmap, my design with the YOSYS synthesizer, however I do not succeed simulating the post-synthesis design.

> I would like to add that SDF is partially supported by Icarus Verilog. I did some tests here with Sky130A: https://github.com/mole99/interconnect-tests Unfortunately, no timing checks are [...]

With Icarus Verilog, the `sg13g2_stdcell.v` file can indeed be read, however the internal register behavior (for instance of `sg13g2_dfrbp_1`) is not correctly interpreted as it relies on the `delayed_CLK` and `delayed_D` signals.
These signals are generated from a timing checks ($setuphold) and are thus not driven by the simulator.
```log
[...]/ihp130/pdk/ihp-sg13g2/libs.ref/sg13g2_stdcell/verilog/sg13g2_stdcell.v:459: warning: Timing checks are not supported and delayed signal "delayed_CLK" will not be driven.
```
Content of `sg13g2_stdcell:l459`:
```verilog
$setuphold (posedge CLK, posedge D, 0, 0, notifier,,, delayed_CLK, delayed_D);
```
As a result, the signal are constantly set to `x` and post-synthesis simulation are not possible.
@mole99 do you know if there is a way to tell Icarus Verilog to ignore the timing checks and drive the output signal as original the source?
If not, it would be nice to have an alternative "pure" RTL description (no timing) of the IHP130 standard for quick functional verification of synthesized netlist.

Blebowski, 2024-06-26:
@MaxenceBouvier would it be OK to add `ifdef FUNCTIONAL` to conditionally compile without timing checks ?

```
module sg13g2_a21oi_1 (Y, A1, A2, B1);
	output Y;
	input A1, A2, B1;

	// Function
	wire int_fwire_0, int_fwire_1;

	and (int_fwire_0, A1, A2);
	or (int_fwire_1, int_fwire_0, B1);
	not (Y, int_fwire_1);

`ifdef FUNCTIONAL
	// Timing
	specify
		(A1 => Y) = 0;
		(A2 => Y) = 0;
		if (A1 == 1'b1 & A2 == 1'b0)
			(B1 => Y) = 0;
		if (A1 == 1'b0 & A2 == 1'b1)
			(B1 => Y) = 0;
		if (A1 == 1'b0 & A2 == 1'b0)
			(B1 => Y) = 0;
		ifnone (B1 => Y) = 0;
	endspecify
`endif
endmodule
`endcelldefine
```

mole99, 2024-06-26:
@MaxenceBouvier You must be using an older version of Icarus Verilog, in the latest version these signals are driven (but not delayed), which should be enough for a GL simulation :)
Can you send me your version string?

MaxenceBouvier, 2024-07-02:
> @MaxenceBouvier You must be using an older version of Icarus Verilog, in the latest version these signals are driven (but not delayed), which should be enough for a GL simulation :) Can you send me your version string?

Indeed, I am checking out version 12.0 (from dec. 2022) before building from source.
I'll point to latest passing commit and try again.
Thanks for letting me know that it should now be working!

mole99, 2024-07-03:
You are welcome @MaxenceBouvier! If there are any problems, let me know.

I agree with you, @Blebowski, that an `ifdef` around the specify blocks could be useful. But you would need to connect the delayed signals in the Flip-Flops.

LucaMartis00, 2024-10-07:
@MaxenceBouvier I have the same problem as you. I want to simulate a post-synthesis netlist, but Icarus Verilog (Iverilog) gives me this warning:
`Timing checks are not supported' and 'delayed signal. "delayed_CLK" will not be driven`
Did you manage to solve it? How did you solve it?

mole99, 2024-10-07:
@LucaMartis00 You need to update Icarus Verilog. (Possibly build from source.)

LucaMartis00, 2024-10-07:
@mole99 I'm using version 13
[image]