IHP-GmbH/IHP-Open-PDK #149: Statistical modeling with parameter mc_ok

miesli, 2024-06-25:
## Expected Behavior

Turning on and off statistical modeling by choosing different corners

## Actual Behavior

Turning on and off statistical modeling by choosing different corners PLUS option to change the sigma of the Gaussian function using mc_ok.

## Fix

There are multiple options depending on the wanted behavior. The most obvious is:

Replace all mc_ok appearances with 1.0 in the ngspice/models/cornerXXX.lib files

### Thanks

I report this here after someone else reported this to me.

dwarning, 2024-06-25:
No, this is partly a misunderstanding:
In a former version of the PDK lib with different library concept mc_ok was used to switch on/off statistical simulation with Monte Carlo.
Actual the selection followed by mos_tt_stat library parameter. Everything works well as we can see in attached picture:
[image]
The problem is that the name mc_ok is not correct anymore. He has now the meaning of Sigma.
The simulation is made with Sigma=6.

simi1505, 2025-01-30:
@dwarning Can you share your testbench or at least how you invoke this corner with Sigma=6?

`.lib cornerMOSlv.lib mos_tt_stat ...?`

I want to try the Monte Carlo simulation in combination with CACE. 

What is a suitable value for Sigma = mc_ok? Where do you get 6 from?

Thanks!

dwarning, 2025-01-30:
Please at first read the ngspice manual (or hspice/spectre) about the meaning of the gauss() and agauss() function parameters.
Assuming the standard deviation values in IHP statistical libraries (I am not sure about this!) are standing for 3 Sigma. Then you get 3 Sigma deviations with mc_ok = 1. Widening the spread to 6 Sigma needs setting mc_ok to 0.5.
A reworked corner/mismatch library concept is pending: https://github.com/IHP-GmbH/IHP-Open-PDK/pull/333
Would be interesting to see spec compliance of a practical circuit with this high deviations.

sergeiandreyev, 2025-01-30:
> Assuming the standard deviation values in IHP statistical libraries (I am not sure about this!) are standing for 3 Sigma.

Yes, these are 3-Sigma

simi1505, 2025-01-30:
Thanks for the fast support!

However, there is something I do not understand. I thought mc_ok equals sigma?

The Ngspice manual defines gauss() as follows:

[image]

`define gauss(nom, rvar, sig) (nom + (nom*rvar)/sig * sgauss(0))`

And in the sg13g2_moslv_stat.lib file it is called with `.param sg13g2_lv_nmos_vfbo_mm  ='gauss(sg13g2_lv_nmos_vfbo_mm_norm , 0.004, mc_ok)'`.

If I now define `.param mc_ok = 3`, I should get `sigma = 3`, right? I do not understand, where you get mc_ok = 1 --> sigma = 3 and mc_ok = 0.5 --> sigma = 6?

hpretl, 2025-02-06:
@dwarning Thanks to the thread above, I finally understand how the `mc_ok` parameter works. One observation, though: Setting `.param mc_ok=0` should lead to divide-by-zero or a crash of ngspice, right? In practice, it seem to turn off the Monte-Carlo variation, at least this is my guess.

Question: Is this what is happening in the ngspice code?

dwarning, 2025-02-06:
@hpretl 
You are correct, if mc_ok = 0 turns off variation and returns nominal value.

Excerpt from inp.c:

`static double
agauss(double nominal_val, double abs_variation, double sigma)
{
    double stdvar;
    if (abs_variation <= 0 || sigma <= 0)
        return nominal_val;
    stdvar = abs_variation / sigma;
    return (nominal_val + stdvar * gauss1());
}

`

hpretl, 2025-02-06:
@dwarning Excellent, so `mc_ok` is indeed a nice MC switch :-)

MC on: `mc_ok=1`
MC off: `mc_ok=0`

sergeiandreyev, 2025-11-05:
Hi @miesli, can we mark this issue as resolved?

miesli, 2025-11-07:
yes, it is resolved. Thanks for the reminder