IHP-GmbH/IHP-Open-PDK #153: qucs-s simulation with xyce

FhatuL, 2024-06-28:
Hi,

I wanted to ask, regarding the qucs-s HBT models, is it possible to run the simulations using xyce instead of ngspice? Every time I try to use the xyce simulator i get this error -> Netlist error: Number of nodes for subcircuit instance XNPN13G2L1 does not
 agree with number of nodes in subcircuit
 IHP_PDK_NONLINEAR_COMPONENTS_NPN13G2L

FhatuL, 2024-07-01:
Hi @KrzysztofHerman , please let me know if i missed something:

- Qucs Version: 24.2.1
- OS/Platform: Ubuntu 24.04

## Expected Behavior
While using the HBT models, I am expecting the simulation to run without errors when using xyce

## Actual Behavior
While trying to run a simulation with xyce, netlist errors occur.
image of error:
[image]

## Steps to Reproduce the Problem

1. Create a schematic and use one of the HBT models, in this case I used the npn13G2l model.
2. Add an include script for the HBT models, in my case, I just copied the include syntax provided in one of the example schematics for qucs-s in the repository.
3. Make sure xyce is selected as the simulator
4. Insert a simulation type in the schematic, for my case I was using the S parameter simulation
5. Try to run the simulation
Image of my schematic:
[image]

KrzysztofHerman, 2024-07-01:
@FhatuL does it run when you use ngspice ?
Anyway QUCS-S only generate the netlist and then calls the simulator.

FhatuL, 2024-07-01:
@KrzysztofHerman It runs when I use ngspice yes.

KrzysztofHerman, 2024-07-01:
Could you please provide the netlist an message log. You can generate it pressing F5 and F6

FhatuL, 2024-07-01:
find attached
[netlist1.txt](https://github.com/user-attachments/files/16051219/netlist1.txt)
[log1.txt](https://github.com/user-attachments/files/16051218/log1.txt)

sergeiandreyev, 2024-07-01:
based on the log it seems that all the `.parameters` commands in `cornerHBT.lib` file are not supported in Xyce, so the parameters are not set and simulation of course could not run
we need to change the syntax of the models in a way to be supported by Xyce (like we did for diodes and resistors models)

sergeiandreyev, 2024-07-01:
ok, after the change `.parameters` to `.param` (Xyce syntax) I'm still getting the 'incorrect number of nodes' error

sergeiandreyev, 2024-07-01:
ok, found it
`.SUBCKT IHP_PDK_nonlinear_components_npn13G2l  gnd c b e bn t Nx=1 El=1 `
here `gnd` is the extra pin

sergeiandreyev, 2024-07-01:
or the other way around, you should add additional pin to the line 10:
`Xnpn13G2l1 0  _net0 _net1 0 gnd IHP_PDK_nonlinear_components_npn13G2l Nx=1 El=1`

sergeiandreyev, 2024-07-01:
so now I'm stuck with this error:
`Model is required for device QNPN13G2L and no valid model card found.`
I'm using Xyce v7.6

FhatuL, 2024-07-02:
> ok, found it `.SUBCKT IHP_PDK_nonlinear_components_npn13G2l gnd c b e bn t Nx=1 El=1 ` here `gnd` is the extra pin

is it the circuit symbol that causes this extra pin to appear? How could i remove it since the netlist is auto-generated in my case.

KrzysztofHerman, 2024-07-02:
in the line 259 of [this](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.tech/qucs/user_lib/IHP_PDK_nonlinear_components.lib) file you have this definition: `.SUBCKT IHP_PDK_nonlinear_components_npn13G2l  gnd c b e bn t Nx=1 El=1 `. You could try to remove `gnd`, which seems to be optional,  and let us know how it goes.

sergeiandreyev, 2024-07-02:
> is it the circuit symbol that causes this extra pin to appear? How could i remove it since the netlist is auto-generated in my case.

I think it comes from this file -> [link](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.tech/qucs/user_lib/IHP_PDK_nonlinear_components.lib)
lines 259-261

sergeiandreyev, 2024-07-03:
thank you, @dwarning !
now at least Xyce is running on my side (standalone)

dwarning, 2024-07-03:
The problem with the nested loop arises after a copy from ngspice schematic. Removing the both parameter sweeps, save and re-insert again solved the problem.

FhatuL, 2024-07-03:
> Just a copy of the 3 hbt files into the xyce model folder with these changes is sufficient. But the netlister seems not to fit xyce requirements for this nested loop: ![Bildschirmfoto vom 2024-07-03 14-34-50](https://private-user-images.githubusercontent.com/18441932/345423163-a141e576-2135-4848-92cc-dd77c65bd70a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAwMjA4MDUsIm5iZiI6MTcyMDAyMDUwNSwicGF0aCI6Ii8xODQ0MTkzMi8zNDU0MjMxNjMtYTE0MWU1NzYtMjEzNS00ODQ4LTkyY2MtZGQ3N2M2NWJkNzBhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAzVDE1MjgyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY0YzJmZjc0NTVlY2Y3ZDhlY2ZjOTk3MDQ0YjdkNWZhYWIzNDllMTNhM2JkYTM3MWVhNzUyZTI4NDlhMGI1OGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8c_0g99TIe9TCwa54Ou2rtCIegfeNWcsKk4ye0nPECw)

Thank you, doing what you said got it to work. I wanted to ask here, for the symbol in this schematic, how did you do the model = npn13G2 nx=8 etc.

FhatuL, 2024-07-03:
Also, removing the extra gnd port on line 259 in the IHP_PDK_nonlinear_components.lib, although it does allow the simulation to run, it gives totally different simulation results compared to when it is there with ngspice. For xyce I just added an extra .PortSym and that seems to have gotten rid of the Number of nodes for subcircuit instance XNPN13G2L2 does not
 agree with number of nodes in subcircuit error.

sergeiandreyev, 2024-07-04:
@FhatuL, great! Could you please attach the final updated netlist here for reference?

FhatuL, 2024-07-04:
> @FhatuL, great! Could you please attach the final updated netlist here for reference? On PDK side, we will provide model files with required changes in `libs.tech/xyce/models` folder

* Qucs 24.2.1  /home/fhatu/test2.sch

.SUBCKT IHP_PDK_nonlinear_components_npn13G2l  gnd c b e bn t Nx=1 El=1 
.INCLUDE   ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/xyce/models/sg13g2_hbt_mod.lib
X1 c b e bn t  npn13G2l Nx={Nx} El={El} 
.ENDS
  
.LIB ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/xyce/models/cornerHBT.lib hbt_typ

Xnpn13G2l1 0  _net0 _net1 0 gnd 0 IHP_PDK_nonlinear_components_npn13G2l Nx=1 El=1
C1 _net2 _net1  1N 
C2 _net0 _net3  1N 
L1 _net0 _net4  1M 
L2 _net5 _net1  1M 
I1 0 _net5 DC 2U
P1 _net2 0 port=1  z0=50  AC 0.316228  SIN 0 0.316228 1MEG
P2 _net3 0 port=2  z0=50  AC 0.316228  SIN 0 0.316228 1MEG

V1 _net4 0 DC 1
.AC LIN 201 10G 30G
.LIN format=touchstone sparcalc=1
.PRINT ac format=std file=spice4qucs_sparam.prn sdb(1,1) s(1,1) sp(1,1) y(1,1) z(1,1) sdb(1,2) s(1,2) sp(1,2) y(1,2) z(1,2) sdb(2,1) s(2,1) sp(2,1) y(2,1) z(2,1) sdb(2,2) s(2,2) sp(2,2) y(2,2) z(2,2)
.END

sergeiandreyev, 2024-07-31:
Hi @FhatuL, we pushed the HBT models for Xyce to the `dev` branch: https://github.com/IHP-GmbH/IHP-Open-PDK/commit/d2c5295c38314a709f2376c70ccd9328f2d99dbf
So we have only the Qucs-S netlisting issue left

Ikram-rs22, 2024-10-19:
@sergeiandreyev I did everything that was mentioned above and I still get this error that no. of nodes don't match. I'm running the dc hbt example that was provided but it runs fine in ngspice after removing gnd pin from npn13G2 in the IHP_nonlinear_component library. But I still get error in xyce simulation. I've downloaded the xyce models, but still it shows the following error in simulation: 
Xyce started...

*****
***** Welcome to the Xyce(TM) Parallel Electronic Simulator
*****
***** This is version Xyce Release 7.6-opensource
***** Date: Sat Oct 19 04:02:13 EDT 2024

***** Executing netlist /home/rf/.cache/qucs-s/spice4qucs.dc.cir

***** Reading and parsing netlist...
Netlist error: Number of nodes for subcircuit instance XNPN13G2 does not agree
 with number of nodes in subcircuit IHP_PDK_NONLINEAR_COMPONENTS_NPN13G2
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 1 MSG_ERROR
 errors
*** Xyce Abort ***
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 1 MSG_ERROR
 errors

*** Xyce Abort ***
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
--------------------------------------------------------------------------
Simulation finished

I'm providing the qucs-s netlist in .cache folder: 
* Qucs 24.3.0  /home/rf/IHP-Open-PDK/ihp-sg13g2/libs.tech/qucs/examples/dc_hbt_13g2.sch

.SUBCKT IHP_PDK_nonlinear_components_npn13G2 gnd c b e bn t Nx=1  
.INCLUDE   ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib
X1 c b e bn t  npn13G2 Nx={Nx} 
.ENDS
  
.LIB /home/rf/IHP-Open-PDK/ihp-sg13g2/libs.tech/xyce/models/cornerHBT.lib hbt_typ
VPr1 _net0 _net1 DC 0
V2 _net0 0 DC 1
I1 0 _net2 DC 0
Xnpn13G2 0  _net1 _net2 0 gnd IHP_PDK_nonlinear_components_npn13G2 Nx=1

.dc v2 0 1.5 0.005
.step i1 0 5e-06 5.55556e-07
.PRINT  dc format=raw file=spice4qucs.dc._swp.plot I(VPr1) 
.END

Please help me. I'm stuck in this problem for almost a month

FhatuL, 2024-10-21:
@Ikram-rs22 hi, what worked for me was to  edit the IHP_PDK_nonlinear_components.lib file, I think this defines the symbols qucs uses. just edit the component you want to use, for example, npn13G2l, first you're going to edit the .INCLUDE to point to the xyce model .lib file, then you're going to add an extra portsym, this <.PortSym 10 0 5 180> works for me and I can simulate with xyce. The downside however is that if you want to use ngspice, you have to redo these edits. hope this helps.

sergeiandreyev, 2024-10-21:
@KrzysztofHerman, could you please take a look?

sergeiandreyev, 2024-10-21:
@Ikram-rs22, could you please double check if you're using the latest PDK `dev` data?
what seems to be inconsistent are the following lines in your netlist:
```
.SUBCKT IHP_PDK_nonlinear_components_npn13G2 gnd c b e bn t Nx=1
.INCLUDE ../../.qucs/IHP-Open-PDK-main/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_hbt_mod.lib
X1 c b e bn t npn13G2 Nx={Nx}
.ENDS
```
This is what we have in `IHP_PDK_nonlinear_components.lib` ([link](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.tech/qucs/user_lib/IHP_PDK_nonlinear_components.lib)):
```
  <Spice>*
.SUBCKT IHP_PDK_nonlinear_components_npn13G2  gnd c b e bn Nx=1  
X1 c b e bn npn13G2 Nx={Nx} 
.ENDS
  </Spice>
```

KrzysztofHerman, 2024-12-18:
@FhatuL @Ikram-rs22 does the latest version of our PDK solves the issues you have, please report ?

Ikram-rs22, 2024-12-18:
No, when using the dev branch both ngspice and xyce are not able to simulate. Therefore, I'm using the main branch where only ngspice works

KrzysztofHerman, 2024-12-18:
@Ikram-rs22 could you please share your test case

Ikram-rs22, 2024-12-18:
Ok I'll share it with you as soon as possible  Mr. Herman

Ikram-rs22, 2024-12-19:
I've downloaded the main branch last night and the result is the same. For both qucs and xyce models, the hbt simulations don't run. However, the resistor example runs only with ngspice. I don't know if I'm using the updated pdk or not. 

[image]

[image]

[image]

Ikram-rs22, 2024-12-19:
However, I can simulate the netlist with xyce externally provided in the examples. This is the npn13G2_gain.cir netlist simulated with xyce externally.

[image]

sergeiandreyev, 2024-12-19:
> I've downloaded the main branch last night and the result is the same. ...
@Ikram-rs22, please download the `dev` branch

Ikram-rs22, 2024-12-19:
The dev branch previously did not work for me. Both xyce and ngspice didn't simulate. Has the dev branch been updated?

Ikram-rs22, 2024-12-20:
@KrzysztofHerman @sergeiandreyev The dev branch is working with Xyce. Thank you

closed 2024-12-20 by sergeiandreyev

FhatuL, 2024-12-20:
Hi, I tested out the new dev branch and it seems to work with both ngspice and Xyce, thanks. Although ngspice works, it seems to be producing some errors, I have attached a picture of these errors.
[image]

KrzysztofHerman, 2024-12-20:
@FhatuL please try to run `qucs-s` from the command line. Perhaps you click icon and qucs-s does not source the environmental variables `$PDK_ROOT` and `$PDK` make sure also that a symbolic link `.spiceinit` exists in your home directory and it links to the right file

FhatuL, 2024-12-20:
@KrzysztofHerman what value should $PDK take?

KrzysztofHerman, 2024-12-20:
`$PDK` = `ihp=sh13g2`

The typical configuration of your `.bashrc` should be as follows
```
export PDK_ROOT=/your_installation_pdk_path/IHP-Open-PDK
export PDK=ihp-sg13g2 
export KLAYOUT_PATH="/home/$USER/.klayout:$PDK_ROOT/$PDK/libs.tech/klayout"
export KLAYOUT_HOME=/home/$USER/.klayout 
```

FhatuL, 2024-12-20:
that worked, thank you.