IHP-GmbH/IHP-Open-PDK #157: Cloning the submodule digital fails

hpretl, 2024-07-04:
## Environment

- Klayout Version: n/a
- OS/Platform: n/a

## Expected Behavior

Cloning the GitHub repo (branch `dev`) of the PDK works.

## Actual Behavior

It fails with the following error message:
```
24.49 Branch 'dev' set up to track remote branch 'dev' from 'origin'.
24.49 Switched to a new branch 'dev'
24.58 Submodule 'ihp-sg13g2/libs.tech/digital' (git@github.com:hneemann/IHP130.git) registered for path 'ihp-sg13g2/libs.tech/digital'
24.60 Cloning into '/tmp/ihp/ihp-sg13g2/libs.tech/digital'...
24.95 Host key verification failed.
24.95 fatal: Could not read from remote repository.
```

## Steps to Reproduce the Problem

1. See above, fails at just cloning the PDK repo.

hpretl, 2024-07-04:
See https://github.com/IHP-GmbH/IHP-Open-PDK/commit/2bcc70772bead98b8dbe6d2399d19530a1441d2c

sergeiandreyev, 2024-07-04:
Hi @hpretl , could you please share the `git` command you're using to clone/populate PDK data?

sergeiandreyev, 2024-07-05:
now that we start to include submodules in PDK repository there should be an additional option `--recursive` to run `git clone`:
```
> git clone --recursive -b dev git@github.com:sergeiandreyev/IHP-Open-PDK.git IHP-Open-PDK
...
Submodule path 'ihp-sg13g2/libs.tech/digital': checked out '9a6e7c9d404f9f4334513a24788aa8454918d400'
```

sergeiandreyev, 2024-07-14:
it seems we figured out the root cause:
in `.gitmodules` we have the reference to the `digital` repo in SSH mode: `git@github..`
and this will not work if the user does not have public keys as a method of authorizing on the GitHub - which is not the case for me and Krzysztof, this is why we did not see the issue
I checked if HTTP mode for repo URL in my fork will work for both use cases and apparently there are no issues now
We need to double check this and then I'll make a PR.

hpretl, 2024-07-14:
That sounds good! During image build there are no user credentials, we use https to pull stuff from GH.

hpretl, 2024-07-16:
Works now, thanks.

PR #159 (2024-07-16): Update .gitmodules
changed SSH to HTTPS mode

Fixes #157 

- [x] Tests pass
closed 2024-07-16 by hpretl