IHP-GmbH/IHP-Open-PDK #158: MOS Devices Tests Aren't Working.

AhmedLilah, 2024-07-14:
## Environment

- Klayout Version: N/A
- OS/Platform: Ubuntu

## Expected Behavior
provided MOS tests to run.

## Actual Behavior
when I run any of the tests in xschem for the most devices I get an error while running ngspice saying 
```plain 
Error on line: 
< the instance line>
unknown device type - error
        Simulation interrupted due to error!
```
this is not the case with passives and BJTs it only happens with most devices the rest are working fine.

## Steps to Reproduce the Problem

1. install the ihp-sg13g2 PDK using the make files and the python script in the xschem folder.
2. copy the ihp-sg13g2 PDK to the PDK_ROOT directory ( in my case /usr/local/share/pdk/)
3. copy the xschemrc file in xschme to my test directory
4. run `xschem` 
5. open the test mos test.
6. netlist and run.

KrzysztofHerman, 2024-07-15:
@AhmedLilah thank you for reporting this issue. 
Our MOS devices use PSP103 model, which is included in ngspice `control` section as a binary file using the following command: 'pre_osdi ./psp103_nq.osdi' Because of the license agreement we can not distribute the `osdi` binary but we do redistribute source code of the PSP103 model. The Verilog-A code can be compiled using opencaf (the `install.py` script does it). The potential issue is that since your `PDK_ROOT` points to the `/usr/local/share/pdk` running `install.py` does not take effect. Try first export the `PDK_ROOT` to point to the actual place where you clone it,  execute `install.py`. It is mandatory to have `openvaf` installed.

KrzysztofHerman, 2024-08-05:
@AhmedLilah was the above comment helpful ?

AhmedLilah, 2024-08-05:
Unfortunately no, I installed it according to what you said and it still doesn't work.

KrzysztofHerman, 2024-08-05:
Did you try our latest updates, where `.spiceinit` is used to let the simulator know about the models and the `osdi` file ?
If no try to update the `dev` branch and run `install.py` once more.

AhmedLilah, 2024-08-05:
I did and got this 

[image]

KrzysztofHerman, 2024-08-06:
1. Make sure that you have exported the env variables
```
export PDK_ROOT=/home/$USER/your_directory/IHP-Open-PDK
export PDK=ihp-sg13g2 
export KLAYOUT_PATH="/home/$USER/.klayout:$PDK_ROOT/$PDK/libs.tech/klayout"
export KLAYOUT_HOME=/home/$USER/.klayout 
```
Note that the `klayout` related stuff is not used in the simulation but you will need this when making layout.

2.  after running `$PDK_ROOT/$PDK/libs.tech/xschem/install.py` make sure that `.spiceinit` symbolic link is created in `/home/$USER` and it points to the `$PDK_ROOT/$PDK/libs.tech/ngspice/.spiceinit`file
3. make sure that the file `$PDK_ROOT/$PDK/libs.tech/ngspice/openvaf/psp103_nqs.osdi` exists
4. run xschem in `$PDK_ROOT/$PDK/libs.tech/xschem` directory

Alternatively you can run this netlist directly in `ngspice`
```
*
.lib cornerMOSlv.lib mos_tt
Vgs net1 GND 0.4
Vds net3 GND 1.0
Vd net3 net2 0
.param temp=27
XM1 net2 net1 GND GND sg13_lv_nmos w=1.0u l=0.13u ng=1 m=1
.control
save all
op
let Id = @n.xm1.nsg13_lv_nmos[ids] 
print Id
.endc
.GLOBAL GND
.end
```

KrzysztofHerman, 2024-08-21:
Can we close the issue ?

sergeiandreyev, 2024-09-11:
Hi @AhmedLilah, do you have some feedback or additional questions on this item? Otherwise we are going to close the issue as solved.

AhmedLilah, 2024-09-11:
@sergeiandreyev 

I will revisit the problem once again and get back if anything changed.

But based on the bast comment it didn't get solved after setting the enviroment variables and compiling the files.

KrzysztofHerman, 2024-10-29:
@AhmedLilah can we close this issue ?

closed 2024-12-18 by KrzysztofHerman