IHP-GmbH/IHP-Open-PDK #166: Routing issue in `dfrbp_1` flop

sergeiandreyev, 2024-07-23:
## Environment

- Klayout Version: 0.29.2
- OS/Platform: Linux

## Expected Behavior
One should be able to connect CLK pin in `dfrbp_1` stdcell flop with M2

## Actual Behavior
DRC violation when the Via is put because of too small distance between Reset M2 bar and CLK pin

## Steps to Reproduce the Problem

1. Try to put min Via on CLK pin
2. Run DRC
3. It will produce violation

KrzysztofHerman, 2024-08-05:
@sergeiandreyev in order to make a valid contact one have to put the via and metal2 on the limit of the technology (see DRC rules `M2.b` and `Via1.c`). 
[image]

The tools like OpenROAD already deal with it without any issues.

[image]

sergeiandreyev, 2024-08-05:
the issue relates to commercial PDK flow

closed 2024-08-05 by sergeiandreyev