IHP-GmbH/IHP-Open-PDK #169: Any Recomended Python Versions?

redpanda3, 2024-07-29:
## Environment

- Klayout Version: .26.2
- OS/Platform: ubuntu20
python version: 3.8

## Expected Behavior

When klayout has launched, there should be no error in the terminal.

## Actual Behavior

ERROR message in the terminal ./.klayout/python/cni/ulist.py:23: "type" object is not subscriptable.

## Steps to Reproduce the Problem

1.
2.
3.

d-m-bailey, 2024-07-29:
@redpanda3 I think the LVS scripts for IHP require at least klayout 0.29.0

redpanda3, 2024-07-29:
> @redpanda3 I think the LVS scripts for IHP require at least klayout 0.29.0

Hi @d-m-bailey,

I've just upgraded that to 0.29.4, but it still shows the error message.

redpanda3, 2024-07-29:
Switch to python3.12.4, still shows the error message.

KrzysztofHerman, 2024-07-29:
@redpanda3 no issues with `python 3.10.12` and `klayout 0.29.0`. The `IHP-Open-PDK` is being developed using `Ubuntu 22.04 LTS`. We do not support backward compatibility with the previous versions of Ubuntu.

d-m-bailey, 2024-07-29:
@redpanda3 can you try `pip install klayout --upgrade`?
I noticed that the python libraries for klayout seemed to be installed differently than the actual program.

sergeiandreyev, 2024-09-11:
Hi @redpanda3, do you have some feedback or additional questions on this item?

redpanda3, 2024-09-12:
[image]
[image]

Hi, @sergeiandreyev @d-m-bailey @KrzysztofHerman please have a check on whether there is a Tech class under cni/dlo.py. I guess the problem is located in here.  Current Klayout version is 0.29.0, python version is 3.10.

redpanda3, 2024-09-12:
Maybe this is the new problem. Haven't check the problem for a month, sorry for that.

sergeiandreyev, 2024-09-13:
we've made un update to fix the `Tech` class import definition in `__init__.py`:
https://github.com/IHP-GmbH/IHP-Open-PDK/pull/200

redpanda3, 2024-09-13:
@sergeiandreyev Thank you, the problem solved, current python is 3.10 with klayout 0.29, ubuntu20.