IHP-GmbH/IHP-Open-PDK #177: sram: missing .lib files

dnltz, 2024-08-12:
Missing `.lib` files for 

- RM_IHPSG13_1P_1024x16_c2_bm_bist
- RM_IHPSG13_1P_1024x8_c2_bm_bist
- RM_IHPSG13_1P_4096x16_c3_bm_bist

verilog, lef, gds, and cdl exist.

sergeiandreyev, 2024-08-12:
Thank you, @dnltz! It's indeed strange, I have the new `.libs` in the standard folder..
`git push` was working w/o any issues, and I have a clean `git status` -> "nothing to commit"
but there is nothing uploaded to the repo
do you know how to debug this?

dnltz, 2024-08-12:
@sergeiandreyev I wasn't aware gitignore also ignores sub-directories ... did a quick test and was able to commit files in `lib/` again.

dnltz, 2024-08-12:
Maybe you can try if #178 allows you to commit the missing libs.

sergeiandreyev, 2024-08-12:
thank you, it works! btw, what was the data that should be skipped w/ `lib` pattern?

dnltz, 2024-08-12:
Perfect. That's where it's defined: https://github.com/IHP-GmbH/IHP-Open-PDK/blob/5a42d03194e8c98558f4e34538338a60550f89b9/.gitignore#L17

I guess it was fine before when it was called `liberty/`.

closed 2024-08-12 by dnltz