IHP-GmbH/IHP-Open-PDK #180: Stdcell schematics should contain multiplier = 1

sergeiandreyev, 2024-08-12:
Layouts contain multi-fingered MOS devices, so CDL and SPICE netlists should be fixed to align with layout

## Expected Behavior
```
.SUBCKT sg13g2_inv_2 A VDD VSS Y
*.PININFO A:I Y:O VDD:B VSS:B
MX1 Y A VSS VSS sg13_lv_nmos m=1 w=1.48u l=130.00n ng=2
MX0 Y A VDD VDD sg13_lv_pmos m=1 w=2.24u l=130.00n ng=2
.ENDS
```
## Actual Behavior
```
.SUBCKT sg13g2_inv_2 A VDD VSS Y
*.PININFO A:I Y:O VDD:B VSS:B
MX1 Y A VSS VSS sg13_lv_nmos m=2 w=740.00n l=130.00n ng=1
MX0 Y A VDD VDD sg13_lv_pmos m=2 w=1.12u l=130.00n ng=1
.ENDS
```

There is a check uploaded: https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.qa/stdcells/check_multiplier.sh

MilosMitrovic411, 2025-02-07:
I just made similar comment on a different issue, but I think it is worth pointing it out here as well. To me it is more natural for "ng" to be a multiplier, not a divisor, in which case your code would look like this:

`.SUBCKT sg13g2_inv_2 A VDD VSS Y
*.PININFO A:I Y:O VDD:B VSS:B
MX1 Y A VSS VSS sg13_lv_nmos m=1 w=740.00n l=130.00n ng=2
MX0 Y A VDD VDD sg13_lv_pmos m=1 w=1.12u l=130.00n ng=2
.ENDS`

sergeiandreyev, 2025-08-15:
verified: ok