IHP-GmbH/IHP-Open-PDK #186: large increase in area for ff compared to buf for upsize

maliberty, 2024-08-19:
@precisionmoon noticed this large jump in area when upsizing a sequential compared to a buffer.  Is this expected?

PDK | BUFX1 | BUFX2 | BUFX1 -> BUFX2 area increase | DFFX1 | DFFX2 | DFFX1 -> DFFX2 area increase | DFF1->2 /BFF1->2
-- | -- | -- | -- | -- | -- | -- | --
ihp | sg13g2_buf_1 | sg13g2_buf_2 |   | sg13g2_dfrbp_1 | sg13g2_dfrbp_2
|   |  7.2576 | 9.072 | 1.8144 | 47.1744 | 54.432 | 7.2576 | 4

KrzysztofHerman, 2024-08-20:
@maliberty thank you for reporting this issue. At a first glance it should not be the case since as shown in the table (I also checked it in LEF and LIB files) the areas of the cells does not increase much as you change the cell type. I have also checked the spice netlist of all 4 circuits and they have identically sized transistors at the output, what should result in equal drive strengths for buffers and FF with the same postfix. I suppose that the increase in the area produces while placing the cells. Could you please mention the most critical parameters which are driving placer/resizer  ?

maliberty, 2024-08-20:
Drive strength, pin capacitance, and area are the most critical.  In other PDKs we observe an smaller ratio of increase.  To me this suggests some inefficiency in the cell layout.

closed 2026-03-05 by maliberty