IHP-GmbH/IHP-Open-PDK #187: PyPreprocessor & new python version

KrzysztofHerman, 2024-08-27:
## Environment

- Klayout Version: 0.29.0
- OS/Platform: Ubuntu 24.04

## Expected Behavior
run klayout without issues

## Actual Behavior
`pypreprocessor` API reports the lack of module imp 

## Steps to Reproduce the Problem

1. Install ubuntu 24.04
2. Compile klayout from sources
3. Clone IHP-Open-PDK
4. Export KLAYOUT_PATH & KLAYOUT_HOME
5. Run klayout

The issue can be solved changing `$PDK_ROOT/$PDK/libs.tech/klayout/python/pypreprocessor/pypreprocessor/__init__.py`
line 11
`import imp`  to `import importlib`

sergeiandreyev, 2024-08-29:
could you please specify Python version that produces the error

adrienluitot, 2024-12-09:
I have the same error from python 3.12, according to several forums the package `imp` seems to be deprecated from python 3.12 indeed.

Currently there are both `import imp` and `import importlib` in the file `pypreprocessor/__init__.py`, but `imp` seems to take the priority.
So I think it is better to completely remove `imp`, as `importlib` seems compatible, stable and usable from older versions too.

closed 2024-12-18 by KrzysztofHerman