IHP-GmbH/IHP-Open-PDK #188: PCells in KLayout

olisnr, 2024-08-28:
i try to use nmosHV, pmosHV and cmim in KLayout from the last working SG13_dev version.

my BiG question is:
1. the MiM-capacitor should only use layer 5 and some MiM-layers if i look at _1.1 Main Processing Sequence and Cross-Section Schematic_ of _**SG13G2 Process Specification Rev. 1.2**_. but in KLayout i see also `TopMetal1.drawing`. is this an error or is it not possible to place a big capacitor over a circuit?

in DRC i get some errors:
2. pmosHV gives minimum errors DRC for pSD.i1 min 400nm, actual 340nm.
3. pmosHV has no GatPloy.pin but nmosHV has. is GatPloy.pin needed for production?
4. if i use an w=2u l=1u pmosHV i get also NW.d1 and LU.a errors
will this be fixed, or is the idea to draw pSD and ThickGateOxyde in the end from hand, when the FETs are placed?

sergeiandreyev, 2024-08-29:
I confirm that `pmosHV` Pycell is not consistent with the implementation in our commercial PDK, the device code will be updated & fixed
hopefully the errors will go away
on the first item let me check on this internally, for now I don't see any difference between PDKs

sergeiandreyev, 2024-08-29:
> 4. if i use an w=2u l=1u pmosHV i get also NW.d1 and LU.a errors

I cannot reproduce this, with the mentioned values I don't have `NW.d1` violations, also `LU.a` errors are present in any case, i.e. w=0.3u l=0.4u (default, min), w=2u l=1u

sergeiandreyev, 2024-08-29:
on the first item, the device geometry is correct - we need to have one connection to Metal5 and another one to TopMetal1
maybe the picture is not clear enough

sergeiandreyev, 2024-09-01:
Hi @olisnr, could you please take a look at the latest commit w/ updated device?
https://github.com/IHP-GmbH/IHP-Open-PDK/commit/017ca57a15805b68717c54ca454ac780541d695c
p.s. `LU.a` error in this device I see also in commercial PDK, I guess it's expected

olisnr, 2024-09-02:
when i use the latest commit (and also some versions before) KLayout (0.29.5 2024-7-31) schows errors at start up:
```
ich@P43s:~$ KLAYOUT_HOME='/home/ich/Downloads/IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/klayout' klayout -e
ERROR: Reading /home/ich/Downloads/IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp/pypreprocessor.py: Unable to open file: /ALL/DoWN/IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp//pypreprocessor.py (errno=2)
ERROR: Reading /home/ich/Downloads/IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp/pypreprocessor.py: Unable to open file: /ALL/DoWN/IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/ihp//pypreprocessor.py (errno=2)
ERROR: /ALL/DoWN/IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/__init__.py:23: ModuleNotFoundError: No module named 'cni'
  /ALL/DoWN/IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/klayout/python/sg13g2_pycell_lib/__init__.py:23
  /ALL/DoWN/IHP-Open-PDK-dev/ihp-sg13g2/libs.tech/klayout/tech/pymacros/autorun.lym:8 (class ModuleNotFoundError)
```

and after it i cant open the `SG13_dev - iHP SG13G2 Pcells`. the window on top is using the newest commit, the window bellow some old version, wihtout the python loading errors:

[image]

sergeiandreyev, 2024-09-02:
I guess we need to update our installation documentation..
https://ihp-open-pdk-docu.readthedocs.io/en/latest/install/installation.html
now that we started to use submodules in the OpenPDK repo there should be additional options/commands to download the PDK correctly:
`git clone -b dev --recursive ...`
will get the branch `dev` with recursive submodules population

olisnr, 2024-09-03:
thanks, now all is totally different :) 

now i get an M1.d error for the pmosHV at default settings. and for all pmosHV the normal LU.a error. 

i tryed to understand the latchup rules. but _Figure 7.4: Latch-up protection rules_  **_SG13G2 Layout Rules Rev. 0.3_** is not so clear for me. 
does the LU.a error means, that one should make a well-contact and tie to VSS?

if i understand it right, thats only necessary in IO cells. 

the other thing i would like to know. how should i isolate sensitive circuits? whit a ring of `N+ isolator`, `P+ isolator` or both (_Figure 7.3: I/O latch-up protection scheme_)?

sergeiandreyev, 2024-09-09:
> I guess we need to update our installation documentation.. https://ihp-open-pdk-docu.readthedocs.io/en/latest/install/installation.html now that we started to use submodules in the OpenPDK repo [...]

updated documentation:
https://ihp-open-pdk-docs.readthedocs.io/en/latest/install/installation.html

sergeiandreyev, 2024-09-09:
> now i get an M1.d error for the pmosHV at default settings. and for all pmosHV the normal LU.a error.

this should be ok, as it is consistent with our commercial PDK

on the latch-up rules - we're reviewing the questions..

closed 2024-11-28 by olisnr