IHP-GmbH/IHP-Open-PDK #191: Resistor parameter `b` usage?

hpretl, 2024-09-02:
The poly resistor has parameters `m` and `b`. `m` is likely multiplicity, e.g. how many resistors in parallel, but what is `b`?

[image]

sergeiandreyev, 2024-09-02:
`b` is number of bends
[image]
here `b=2`

hpretl, 2024-09-02:
Thanks! So no parameter for number of pieces in series (I am usually not using bends).

sergeiandreyev, 2024-09-02:
no, maybe we could consider this as an enhancement, but we need to discuss this internally to be aligned with commercial PDK

sergeiandreyev, 2024-09-18:
Hi @hpretl, after talking it over, we decided not to go ahead with series implementation

hpretl, 2024-09-18:
OK, then let's close this.

closed 2024-09-18 by sergeiandreyev