IHP-GmbH/IHP-Open-PDK #192: top_lvl_pin

olisnr, 2024-09-03:
how are `top-level-pins ` made in KLayout? 
there are `MetalN.pin`, `TopMetalN.pin` and `GatePoly.pin` layers. but there are also the `*.text` layers.
is a top_lvl_pin for the LVS-script a combination of `*.pin` and `*.text`, or something different?

FaragElsayed2, 2024-09-06:
@olisnr 

Text layer only

https://github.com/IHP-GmbH/IHP-Open-PDK/blob/5a42d03194e8c98558f4e34538338a60550f89b9/ihp-sg13g2/libs.tech/klayout/tech/lvs/rule_decks/general_connections.lvs#L55

olisnr, 2024-09-06:
@FaragElsayed2

thanks. is this also working for GatPoly?

atorkmabrains, 2024-09-06:
@olisnr No, only metals.

Is that a request for change?

@sergeiandreyev Do you want us to change that and allow adding text on poly layer?

olisnr, 2024-09-06:
@FaragElsayed2 i think for the MOSFETs it would be necessary. otherwise its not possible to route direct via GatPoly...

sergeiandreyev, 2024-09-06:
@atorkmabrains, I'm not sure if there was such request before
let me check on this internally
How it's done in commercial PDK? 
Frank will be next week I hope, so we can discuss from both sides

atorkmabrains, 2024-09-06:
@olisnr Are you using automated routing for MOSFETs? 

It's unconventional in custom design to need text on poly. Usually, you need pins for the ports of your cells/blocks. There might be some use case of supporting this and it's not a big change for us.

@sergeiandreyev I'm not sure how it was implemented in commercial PDK, but we will check and get back to you.

olisnr, 2024-09-06:
@FaragElsayed2 at the moment i make manual routed analog. 
if one does an OTA or some bias-circuit, then it could be interesting to connect direct to the gates, instead to make some vias to a metal-level thats not really needed on the wrong side of the big MOSFET.

FaragElsayed2, 2024-09-06:
@atorkmabrains @sergeiandreyev @olisnr 

BTW, there is no text layer for poly in the PDK.

atorkmabrains, 2024-09-07:
@olisnr Just to confirm, you mean routing using Poly? Poly routing is not recommended. Poly has high resistivity and noisy routing layer. Just from the electronics point of view.

But as you say, we should cater for any kind of designer. Anyhow, we will check how it's done in the Commercial PDK and get back to you.

olisnr, 2024-09-07:
i only want to rout over GatPoly.drawing if two gates are very near. but it would interessting also when i want to route over metal from a big FET. because, if i have a pin on GatPoly.drawing i can place the Cont.drawing via at the best place, instead of being forced to use predefined vias...

i would like to know, if You think this kind of GatPoly.drawing-routing is a problem:

[OTAtest_small.gds.zip](https://github.com/user-attachments/files/16918351/OTAtest_small.gds.zip)

KrzysztofHerman, 2024-12-18:
@olisnr  can we close this discussion ? The answer to your question was already provided.

closed 2024-12-18 by olisnr

engrvip123, 2025-05-27:
What is the right way for adding pins in layout using IHP PDK in KLayout? Should I draw a metal box over the net using the `.pin` layer and then place pin name as text using same `.pin` layer? 

[image]

[image]

olisnr, 2025-05-27:
@engrvip123 for LVS You have to use a text on the MetlaXXX.text or TopMetalXX.text layer, that is inside a shape of the corresponding .drawing layer. 
i dont know what the pin-layer is good for, may be for the standard-cells and the auto-router...