IHP-GmbH/IHP-Open-PDK #201: DRC seem to miss error on GatPoly Gat.b or Gat.b1

olisnr, 2024-09-16:
KLayout 2.9.6

2 different GatPoly should have 180nm or 250nm distance. but there is no DRC error at 70nm.

im not sure what the correct minimum is in this case:

[image]

akrinke, 2024-09-17:
Thank you for the report! I have created the pull request #204 that should fix your problem. It updates the DRC script files [`sg13g2_maximal.lydrc`](https://raw.githubusercontent.com/IHP-GmbH/IHP-Open-PDK/bb559e2a6d23c66e1788eace1e6033d16964bf05/ihp-sg13g2/libs.tech/klayout/tech/drc/sg13g2_maximal.lydrc) and [`sg13g2_minimal.lydrc`](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/bb559e2a6d23c66e1788eace1e6033d16964bf05/ihp-sg13g2/libs.tech/klayout/tech/drc/sg13g2_minimal.lydrc), so you just have to replace these files.

olisnr, 2024-09-18:
@akrinke the minimal file gives me an error:
```
ich@P43s:~$ KLAYOUT_HOME='/home/ich/share/pdk/ihp-sg13g2/libs.tech/klayout'   klayout -e
ERROR: Reading /ALL/Xschem/ihp/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/drc/sg13g2_minimal.lydrc: XML parser error: Root element must be klayout-macro in line 13, column 3
ERROR: Reading /ALL/Xschem/ihp/IHP-Open-PDK/ihp-sg13g2/libs.tech/klayout/tech/drc/sg13g2_minimal.lydrc: XML parser error: Root element must be klayout-macro in line 13, column 3
```

akrinke, 2024-09-18:
This is strange. The pull request has been merged just recently into the dev branch. The XML header has not been changed. Did you use the latest dev branch?

olisnr, 2024-09-18:
i only replaced the two files. now i try `git clone -b dev --recursive https://github.com/IHP-GmbH/IHP-Open-PDK.git`

olisnr, 2024-09-18:
now its working. 
i would also like an %-free DRC script that ignore all this % fill specifications. because routing-errors would be easier to eliminate, if one wouldnt need to go to check every time 10 filling errors, that show up anyway.

sergeiandreyev, 2024-09-18:
> and are You interested to apply the commercial DRC and LVS tests with a little OTA, that seems to pass the openPDK DRC and LVS?

@olisnr, yes, we can run it

sergeiandreyev, 2024-09-18:
> i would also like an %-free DRC script that ignore all this % fill specifications. because routing-errors would be easier to eliminate, if one wouldnt need to check every time 10 filling errors, that show up anyway.

it will be nice to have different DRC modes under switches (like it's done in OpenPDK LVS), that the user can easily setup from the GUI / cmdline based on what he needs

olisnr, 2024-09-18:
@sergeiandreyev super thanks!

[OTA33.zip](https://github.com/user-attachments/files/17043983/OTA33.zip)

akrinke, 2024-09-18:
I will look into switching DRC modes from the GUI. It should already be possible on the command line:
```
$ klayout -zz -s layout.gds -rd density=false -r sg13g2_minimal.lydrc
```

sergeiandreyev, 2024-09-19:
> @sergeiandreyev super thanks!
> 
> [OTA33.zip](https://github.com/user-attachments/files/17043983/OTA33.zip)

ok, here comes the DRC part..
I have 16 DRC violations in this design:
- Cnt_d: 3
- M1_b: 2
- M1_e: 7
- M2_e: 1
- NW_b: 1
- NW_e: 2

_I'm not sure if I can share more w/o NDA_

olisnr, 2024-09-19:
thanks a lot. 
i think, a print-screens of the error-markers should be ok, or a list of problematic coordinates. the errors are no secret, because the rules are public anyway.

olisnr, 2024-09-19:
and one <180nm metal 1:

[image]

olisnr, 2024-09-19:
metal 1 line > 300nm for >1µm distance > 220nm

[image]

olisnr, 2024-09-19:
Min. space of Metal(n) lines if, at least one line is wider than
0.39 µm and the parallel run is more than 1.0 µm:

[image]

olisnr, 2024-09-19:
not really sure if i got this rule correct, but i think:

Min. NWell enclosure of NWell tie surrounded entirely by NWell in
N+Activ not inside ThickGateOx  0.24

is:

[image]

sergeiandreyev, 2024-09-20:
> the Cnt_d: 3 < 70nm i found already. the openPDK script was missing it:

all these violated rules (apart from `M1_e`) are not yet implemented ([link](
https://github.com/IHP-GmbH/IHP-Open-PDK/blob/dev/ihp-sg13g2/libs.tech/klayout/tech/drc/MissingRules_maximal.md))
@akrinke, do you have a preliminary timeline for the next bunch of DRC rules?

sergeiandreyev, 2024-09-20:
please find attached the DRC violations coordinates:
[drc_OTA.txt](https://github.com/user-attachments/files/17075020/drc_OTA.txt)
how to read:
`8634 -46580` -> `8.634, -46.58`

sergeiandreyev, 2024-09-20:
on the LVS:
- there are absent properties in CDL (`c` for `cmim`)
- discrepancies in `ng` for high-voltage PMOS devices

akrinke, 2024-09-20:
> @akrinke, do you have a preliminary timeline for the next bunch of DRC rules?

We hope to deliver the next version in about two weeks.

sergeiandreyev, 2024-09-23:
> and one <180nm metal 1:
> 
> ![Screenshot from 2024-09-19 21-58-15](https://private-user-images.githubusercontent.com/120017377/369150878-7ebe08a4-c328-47c6-87b5-78e748213ef8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwODg5NTMsIm5iZiI6MTcyNzA4ODY1MywicGF0aCI6Ii8xMjAwMTczNzcvMzY5MTUwODc4LTdlYmUwOGE0LWMzMjgtNDdjNi04N2I1LTc4ZTc0ODIxM2VmOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyM1QxMDUwNTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYjQ3Mzg2MjhjM2QwYTMxYzYzNTViOTU3N2JiMzVmMjkwMGVmZDdlZmYwMDFiOWU5YmZlZjE0NDhhMmJkOTlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.x06xjtMkQM_xb9PVk2CajxGSE7UscgVsEoA_1_MJfVs)

yes, this one is `M1.b`

olisnr, 2024-09-23:
@sergeiandreyev can You run this also on DRC? i tried to fix all DRC errors. and then i have the question if its possible, to run the commercial DRC on layouteditor.de? or on microwaveOffice?
thanks a lot

[OTA_33u 51.gds.zip](https://github.com/user-attachments/files/17097343/OTA_33u.51.gds.zip)

sergeiandreyev, 2024-09-23:
@olisnr, clean! 👍 
> if its possible, to run the commercial DRC on layouteditor.de? or on microwaveOffice?
I guess no, but let me double check..
can you post this question on Slack?

sergeiandreyev, 2024-10-09:
Hi @olisnr, do you have any more questions, or we can close the ticket?

olisnr, 2024-11-28:
sorry

closed 2024-11-28 by olisnr