IHP-GmbH/IHP-Open-PDK #203: how to use tolerances in LVS?

olisnr, 2024-09-17:
i have MOSFETs with 15.975µm instead of 16:
[image]

what is the correct way to define tolerances for part-parameters?

FaragElsayed2, 2024-09-18:
@olisnr Please share a small test case for this. Most probably, it's a grid issue but I need to recheck.

olisnr, 2024-09-18:
@FaragElsayed2 yes, its a grid issue because 16µm/9 dont fit in a 5nm grid. if i change the value in the netlist, it works, but if there are many parts, that need to be changed from hand, it will be interesting to be able solve the problem via a tolerance. the testcase is: $87

[OTA33_grid.zip](https://github.com/user-attachments/files/17045252/OTA33_grid.zip)

FaragElsayed2, 2024-09-18:
@olisnr If you're confident that the issue is grid-related, there's no need to investigate further. To resolve it, you have two options:

    - Fix the DRC grid by trying a few methods, such as adjusting the database unit/scaling.

    - Add a tolerance for the specific parameter you need for each device, similar to the approach outlined here: [KLayout LVS Global Parameters](https://www.klayout.de/doc-qt5/about/lvs_ref_global.html#h2-248). However, this would require modifying the LVS runsets, which is not generally recommended.

By default, we’ve only added tolerances for inductors due to the complexity of geometry calculations.

FaragElsayed2, 2024-09-18:
@olisnr I believe this isn't the ideal use case, but it might be worth discussing internally first.
CC @atorkmabrains.

MilosMitrovic411, 2025-02-07:
I think the good solution would be for "ng" to be a multiplier not a divisor. In your case, when you calculated your total width you would have to divide it yourself and that would immediately tell you that you would have to adjust your value, and plus you would be able to chose yourself do you want your finger width to be 1.775u or 1.78u.

olisnr, 2025-02-08:
@MilosMitrovic411 would be possible, but breaks the compatibility with the commercial PDK...

MilosMitrovic411, 2025-02-08:
@olisnr As it is now, yes, but my point is that it should be changed.