IHP-GmbH/IHP-Open-PDK #205: .spiceinit seems to produce errors in AC simulation

olisnr, 2024-09-18:
the `option savecurrents` from `ihp-sg13g2/libs.tech/ngspice/.spiceinit` produces an error in the ngspice-42 AC-simulation:
`Error: r.x7.r14: Current and power not available for ac analysis`

holvo, 2024-09-24:
Please provide a simple test case showing this issue.

holvo, 2024-09-24:
No test case required. A simple netlist like
```
test ac
.options savecurrents
V1 1 0 dc 0 ac 1
R1 1 2 1k
C1 2 0 1u
.control
ac dec 10 10 10k
display
plot mag(v(2)) mag(@r1[i])
.endc
.end
```
does show this error message. AC simulation and .options savecurrents are incompatible. A resulting vector like @r1[i] is not a complex number and thus does not show magnitude and phase, as is expected by ac results.

KrzysztofHerman, 2024-10-14:
Solved in #217 PR

sergeiandreyev, 2024-10-23:
Hi @olisnr, could you please provide your feedback if the solution works for you and you don't have any additional comments on this issue, after that we can close it

olisnr, 2024-10-23:
looks good. thanks

closed 2024-10-23 by olisnr