IHP-GmbH/IHP-Open-PDK #207: Simulation of MOSFET noise with ngspice

0y8w1x, 2024-09-22:
## Environment

- Klayout Version: not applicable
- OS/Platform: Ubuntu 22.04

## Actual Behavior

I tried to do a noise simulation of a single common source transistor, but found the noise to be 0. In case the schematic is incorrect or I use the noise analysis wrong, please let me know. A simple DC analysis works just fine, so the general setup should be working.

My circuit looks like this:
[image]

And when I plot the onoise_spectrum/inoise_spectrum aswell as the total integrated noise I get the following:
[image]

bmurmann, 2024-09-22:
You are measuring the output noise at a node that is pinned to 1V, so I am not surprised that the result is zero. What are you trying to do?

0y8w1x, 2024-09-29:
> You are measuring the output noise at a node that is pinned to 1V, so I am not surprised that the result is zero. What are you trying to do?

Oh yeah that was a stupid mistake. I want to measure the intrinsic noise of the MOSFET

KrzysztofHerman, 2024-10-09:
@0y8w1x could you please provide a correct testbench with some results. After that we could close the issue with a working example.

dwarning, 2024-10-09:
Maybe these qucs-s/ngspice (NGSPICE_OSDI_DIR env variable have to be set) examples can help to close the issue:
[image]
[image]

[lv_nmos_NOISE_id.sch.txt](https://github.com/user-attachments/files/17311124/lv_nmos_NOISE_id.sch.txt)
[lv_nmos_NOISE_vd_vg.sch.txt](https://github.com/user-attachments/files/17311125/lv_nmos_NOISE_vd_vg.sch.txt)

KrzysztofHerman, 2024-10-10:
@dwarning we provide a sample `.spiceinit`, which sets the location of the OSDI file so you do not have to reference it on the schematic. The only thing you have to do is to create a symbolic link to this file under your `$HOME` direcory. I will also add some more controls to setup the noise plot type.

dwarning, 2024-10-10:
Sorry, it was not the emphasis of my post. Anyway - thanks for hint.
What about induced gate noise. There are data and test benches available?

0y8w1x, 2024-10-12:
thanks @dwarning for the example, I reproduced it and it works great.
[image]

dwarning, 2024-10-14:
Another test case in a 50 Ohm environment NF, NFMIN and Rgn =f(Freq)
[image]

sergeiandreyev, 2024-10-22:
Hi @0y8w1x, do you have additional questions, or we can close the issue?

0y8w1x, 2024-11-21:
> Hi @0y8w1x, do you have additional questions, or we can close the issue?

can be closed, thank you.

closed 2024-11-21 by 0y8w1x