IHP-GmbH/IHP-Open-PDK #208: [bug] LRM specify block delay path restrictions

likeamahoney, 2024-09-23:
Hi, All!

I noticed that in [verilog cell file](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/5a42d03194e8c98558f4e34538338a60550f89b9/ihp-sg13g2/libs.ref/sg13g2_io/verilog/sg13g2_io.v#L75) there are a lot of `specify` constraints which are buggy due to LRM:

> 30.4.1 Module path restrictions
> Module paths have the following restrictions:
> —The module path source shall be a net that is connected to a module input port or inout port.
> —The module path destination shall be a net or variable that is connected to a module output port or
> inout port.

Path can't start at output port due to LRM 2017 and it's earliest versions. So Does it correct to specify such constraints in that manner or it's bug in provided cells?

I use the [`slang`](https://github.com/MikePopoloski/slang/)  tool for elaboration of designs from https://github.com/IHP-GmbH/IHP-Open-PDK/tree/main . With his help I discovered this

sergeiandreyev, 2024-09-30:
Hi @dnltz, could you please take a look at this issue?

dnltz, 2024-09-30:
My fault. Copied the specify block wrongly from the stdcells. I honestly don't know if this correct since timings are just added with 0 delay. At least it shouldn't break any tools for now.

sergeiandreyev, 2024-09-30:
@dnltz, thank you for quick turnaround!

KrzysztofHerman, 2024-12-18:
@likeamahoney does it solve the issue for you, can we close ?

likeamahoney, 2024-12-19:
> @likeamahoney does it solve the issue for you, can we close ?

Yes it is a good enough solution!

PR #215 (2024-09-30): iho-sg13g2: libs.ref: sg13g2_io: verilog: Fix specify syntax
The input and output definition in the Verilog specify block were positioned in the wrong order (output to input). Change and also implement the tri-state better.

Fixes #208
closed 2024-12-19 by likeamahoney