IHP-GmbH/IHP-Open-PDK #209: [bug] ifnone state-dependent path delay

likeamahoney, 2024-09-24:
Hi, All!

I found that `ifnone` condition used [here](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/5a42d03194e8c98558f4e34538338a60550f89b9/ihp-sg13g2/libs.ref/sg13g2_stdcell/verilog/sg13g2_stdcell.v#L1562) and [here](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/5a42d03194e8c98558f4e34538338a60550f89b9/ihp-sg13g2/libs.ref/sg13g2_stdcell/verilog/sg13g2_stdcell.v#L1567) is illegal due to SystemVerilog LRM IEEE 1800-2017 and it's previous versions due to section 30.4.4.4 - The ifnone condition description:

> The ifnone keyword is used to specify a default state-dependent path delay when all other conditions for
> the path are false. The ifnone condition shall specify the same module path source and destination as the
> state-dependent module paths. The following rules apply to module paths specified with the ifnone
> condition:
> —
> Only simple module paths may be described with an ifnone condition.
> —The state-dependent paths that correspond to the ifnone path may be either simple module paths or
> edge-sensitive paths

> —If there are no corresponding state-dependent module paths to the ifnone module path, then the
> ifnone module path shall be treated the same as an unconditional simple module path.
> —
> It is illegal to specify both an ifnone condition for a module path and an unconditional simple
> module path for the same module path.
> 

Which means that edge-sensitive path can't be specified under `ifnone` condition. Edge sensitive paths can be specified only in `if` conditions which correspond to `ifnone` .

Blebowski, 2024-09-30:
Hi @likeamahoney,

maybe I got it wrong, but doesn't the following:
```
The state-dependent paths that correspond to the ifnone path may be either simple module paths or
**edge-sensitive paths**
```
allow having the edge-sensitive paths ?

sergeiandreyev, 2024-09-30:
Hi @Blebowski, there was a thread on this issue:
https://sourceforge.net/p/iverilog/feature-requests/32

Blebowski, 2024-09-30:
Oh, I see, thanks.

It seems like inconsistency in the LRM itself, or not ?

The grammar says:
```
state_dependent_path_declaration ::=
      if ( module_path_expression ) simple_path_declaration
   | if ( module_path_expression ) edge_sensitive_path_declaration
   | ifnone simple_path_declaration
```

but the description says:

```
The state-dependent paths that correspond to the ifnone path may be either simple module paths or
edge-sensitive paths.
```

which would imply that edge-sensitive paths should be allowed here.

Blebowski, 2024-09-30:
Oh, I see now:
```
You are misinterpreting B. It says the paths that correspond to the ifnone may be either simple or edge-sensitive not that the ifnone path can be edge-sensitive. For this definition correspond means the other state-dependent delays that have the same input and output terminals as the ifnone.

FYI this text is basically straight from the standard. 
```

sergeiandreyev, 2024-09-30:
yes, basically we need to figure out how to update our `specify` blocks in Verilog models to be compatible with open source tools (which strictly follow the standard)
we did not catch the issue, because we're using commercial tools for PDK validation, and they have some features on top of the standard as it shows up

b10346, 2026-08-29:
It is solved in the latest standard cell update or not actual at the moment.
Could anybody check the status and confirm that the issue can be closed?

sergeiandreyev, 2026-08-31:
> It is solved in the latest standard cell update or not actual at the moment.
Could anybody check the status and confirm that the issue can be closed?

Hi @likeamahoney, is this issue still valid? (see above)

likeamahoney, 2026-08-31:
> It is solved in the latest standard cell update or not actual at the moment. Could anybody check the status and confirm that the issue can be closed?

> > It is solved in the latest standard cell update or not actual at the moment.
> > Could anybody check the status and confirm that the issue can be closed?
> 
> Hi [@likeamahoney](https://github.com/likeamahoney), is this issue still valid? (see above)

The issue is not resolved.

There are now even more problematic places in the standard cells, for example [here](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/331c00484213b13414777eec1336ef5c29b969bd/ihp-sg13g2/libs.ref/sg13g2_stdcell/verilog/sg13g2_stdcell.v#L42) and [here](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/331c00484213b13414777eec1336ef5c29b969bd/ihp-sg13g2/libs.ref/sg13g2_stdcell/verilog/sg13g2_stdcell.v#L44) (and throughout the rest of the code).

As mentioned by that user, edge-sensitive cells cannot be used in ifnone according to the LRM:

> 30.4.4.4 The ifnone condition
> The ifnone keyword is used to specify a default state-dependent path delay when all other conditions for
> the path are false. The ifnone condition shall specify the same module path source and destination as the
> state-dependent module paths. The following rules apply to module paths specified with the ifnone
> condition:
> — Only simple module paths may be described with an ifnone condition.
> — The state-dependent paths that correspond to the ifnone path may be either simple module paths or
> edge-sensitive paths.
> — If there are no corresponding state-dependent module paths to the ifnone module path, then the
> ifnone module path shall be treated the same as an unconditional simple module path.
> — It is illegal to specify both an ifnone condition for a module path and an unconditional simple
> module path for the same module path.

I also tested this with commercial tools (like VCS and Questa), and unlike the open-source tools, they do accept this construct. However, this still appears to be non-compliant with the LRM.

If supporting the behavior accepted by commercial tools is sufficient for your purposes, then I suppose this issue can be closed. Otherwise, these cases should still be addressed.

b10346, 2026-10-09:
@likeamahoney @Blebowski Could you please take a look at the issue #1228 ?

likeamahoney, 2026-10-09:
> 12\. [@likeamahoney](https://github.com/likeamahoney) [@Blebowski](https://github.com/Blebowski) Could you please take a look at the issue [#1228](https://github.com/IHP-GmbH/IHP-Open-PDK/issues/1228) ?

Thanks for the pointer. https://github.com/IHP-GmbH/IHP-Open-PDK/issues/1228 is related to #209, which is still open: as I noted there, the current sg13g2_stdcell.v uses edge-sensitive paths under ifnone in many cells (e.g. ifnone (posedge CLK => (Q +: D)) = ...), which violates IEEE 1800-2017 §30.4.4.4. The grammar only allows ifnone simple_path_declaration. VCS and Questa accept it, but strict open-source frontends reject it.

If the models are going to be regenerated to add USE_POWER_PINS variants, it would be a good opportunity to fix the specify blocks in the same pass rather than touching the file twice. Possible fixes are converting these to simple paths (ifnone (CLK => Q) = ...) or dropping the ifnone and keeping only the conditional if (...) edge-sensitive paths. Either way, SDF/Liberty consistency (COND/CONDELSE matching) should be double-checked.

It would also make sense to validate the generated models with at least one open-source tool (e.g. Icarus or slang) in CI, so issues like this are caught automatically.

b10346, 2026-10-09:
@likeamahoney could you please review files which are attached to #1228 ? Timing sections of both files are generated from the same Liberty model and consistent. The "_standard".txt version of verilog model does not use edge-sensitive paths with ifnone and uses simple path declaration for combinational cells. Veralator and IVerilog do not show errors about "ifnone" during compilation. We would like to have clearly understanding that format "_standard".txt solves the issue.

Blebowski, 2026-10-09:
Hi,
I don't have strong opinion here. On one side, tools should follow standard. However, if IHP130 follows the standard, there is no pressure on OS tools to adopt the behavior accepted by commercial tools. Slang does a lot in this direction with various compatibility switches. The end result will be that OS tools keep failing to compile commercial PDK cell models since those simply will use the non-standard constructs.

As for the "power pins" topic, since you include power pins already, it would be good to model the functionality. The cell would function correctly only when `vdd == 1 && gnd == 0`. Otherwise, internal state is corrupted (unless being some sort-off retention flip-flop), and outputs are assigned to X. 

This approach allows simulating multi-power domain designs without having UPF flow (which OS simulators do not support).

likeamahoney, 2026-10-09:
> 14\. [@likeamahoney](https://github.com/likeamahoney) could you please review files which are attached to [#1228](https://github.com/IHP-GmbH/IHP-Open-PDK/issues/1228) ? Timing sections of both files are generated from the same Liberty model and consistent. The "_standard".txt version of verilog model does not use edge-sensitive paths with ifnone and uses simple path declaration for combinational cells. Veralator and IVerilog do not show errors about "ifnone" during compilation. We would like to have clearly understanding that format "_standard".txt solves the issue.

I checked both files with slang (which enforces IEEE 1800-2017 §30.4.4.4 strictly), with and without `USE_POWER_PINS`.

The "standard" version fixes almost everything: combinational cells use `ifnone` only with simple paths, and the regular flops use unconditional edge-sensitive paths without `ifnone`, which is fine. However, 8 violations remain in the scan flops `sg13g2_sdfbbp_1`, `sg13g2_sdfrbp_1`, `sg13g2_sdfrbp_2`, `sg13g2_sdfrbpq_1` and `sg13g2_sdfrbpq_2`, e.g.:

```verilog
ifnone (posedge CLK => (Q+:((D && SCD) || ...))) = 0;
```

This is the same construct as in #209 (slang: `ifnone` specify path cannot be an edge-sensitive path). For comparison, the "posneg" version has 68 such errors.

Replacing these lines with simple paths, i.e. `ifnone (CLK => Q) = 0;` and `ifnone (CLK => Q_N) = 0;`, makes the whole file pass slang without errors. This is exactly what the LRM allows: the `ifnone` path is simple, while the corresponding if (...) paths may stay edge-sensitive. Please double-check that SDF annotation still matches these arcs in your flow (`CONDELSE IOPATH` with an edge-qualified input vs. a simple path).

Note that Verilator and Icarus are not a good indicator here: Verilator essentially ignores specify blocks, and Icarus does not fully check them.

So my vote is for the "standard" variant, with the scan flop arcs fixed as above.

b10346, 2026-10-12:
@Blebowski @likeamahoney Thanks for the detailed feedback — we really appreciate it and want to make sure we get this right.

We use automated tools to generate the specify timing section, so it’s not yet clear how to automatically produce simple ifnone paths for scan flops.  Manually editing scan flop timing could introduce errors, so we’d like to be careful.  We’d appreciate a bit more time to review the options and make a final decision.

Blebowski, 2026-10-12:
Hi @b10346 ,

thanks for all the effort. I don't know if the characterizer you use has such option. I would not be surprised if it is not the case. It would make sense that the Simulator vendors adapted to what the characterizers were able to provide.

FYI, I started to write my own characterizer at:
https://github.com/Blebowski/OpenChar

But it is in very raw state, so please don't use it yet (no scan flops, no latches, no clock gates ATM). I would like to spend more time on it (hopefully Q1 next year) and get it into a state where it will be able to characterize IHP130 and SKY130.

likeamahoney, 2026-10-12:
> 17\. [@Blebowski](https://github.com/Blebowski) [@likeamahoney](https://github.com/likeamahoney) Thanks for the detailed feedback — we really appreciate it and want to make sure we get this right.
>      We use automated tools to generate the specify timing section, so it’s not yet clear how to automatically produce simple ifnone paths for scan flops.  Manually editing scan flop timing could introduce errors, so we’d like to be careful.  We’d appreciate a bit more time to review the options and make a final decision.

Thanks, that makes sense. If it helps, I can write a small post-processing script to run after `Liberate`. It would rewrite only edge-sensitive paths under ifnone into simple paths (e.g. `ifnone (posedge CLK => (Q+:...)) = 0; → ifnone (CLK => Q) = 0;`) and keep everything else unchanged. The transformation is deterministic and currently affects only 8 lines in 5 scan flops, so the diff is easy to review. The same script could also run in check-only mode in CI to catch such constructs in future regenerations.

The one thing I can’t verify on my side is SDF back-annotation: for `sdfrbp`/`sdfbbp` the `ifnone` arc is the functional-mode (`SCE=0`) clock-to-Q arc, so it would be good to confirm with your simulators that the `CONDELSE IOPATH` from SDF still annotates to the simple path. Let me know if you’d like me to prepare it.

b10346, 2026-10-12:
@likeamahoney @Blebowski 
Could you please check this version of model?

[sg13g2_stdcell_standard_power_aware.txt](https://github.com/user-attachments/files/33341324/sg13g2_stdcell_standard_power_aware.txt)

likeamahoney, 2026-10-12:
> 20\. [@likeamahoney](https://github.com/likeamahoney) [@Blebowski](https://github.com/Blebowski)
>      Could you please check this version of model?
>      [sg13g2_stdcell_standard_power_aware.txt](https://github.com/user-attachments/files/33341324/sg13g2_stdcell_standard_power_aware.txt)

I checked the new version and the `ifnone` paths are now compliant with IEEE 1800-2017 §30.4.4.4, so from my side #209 can be closed.

Blebowski, 2026-10-13:
Hi @b10346,

thanks, that is what I meant.