IHP-GmbH/IHP-Open-PDK #219: KLAYOUT Unable to open SG13G2.lyp file, Initialization of Layers FAILED

Ikram-rs22, 2024-10-11:
## Environment

- Klayout Version: 0.29.0
- OS/Platform: ubuntu 22

## Expected Behavior

KLayout should load the pycells of the SG13G2 PDK 

## Actual Behavior
We're trying to integrate the PDK with Klayout but after adding the technology files the following error occurs. 
[image]

As a result no nmos or npn library is present. Even if I specify the path of sg13g2_pycell_lib folder in the autorun.lym , it shows the following error
[image]

Can anyone please guide me step by step on the integration process please? I know I'm asking for  a lot but I'm really struggling considering the fact I'm a beginner

## Steps to Reproduce the Problem

1. Klayout opening
2. Adding technology file .lyt from IHP PDK Path using Manage Technology package. Also adding sg13g2.lyp layer files
3.Restarting KLayout

KrzysztofHerman, 2024-10-14:
@Ikram-rs22 thanks for reporting the issue. 
Let me guide you 

1. Clone the repository using `git clone --recursive https://github.com/IHP-GmbH/IHP-Open-PDK.git`
2. Setup PDK_ROOT, PDK, KLAYOUT_PATH and KLAYOUT_HOME in your `.bashrc`  file
```
export PDK_ROOT= your_directory/IHP-Open-PDK
export PDK=ihp-sg13g2 
export KLAYOUT_PATH="/home/$USER/.klayout:$PDK_ROOT/$PDK/libs.tech/klayout"
export KLAYOUT_HOME=/home/$USER/.klayout 
```
`source ~/.bashrc`

3.  Change the branch to `dev` 

```
cd $PDK_ROOT
git checkout dev
```
4.  Run klayout: `klayout -e`

Ikram-rs22, 2024-10-15:
Thanks a lot Mr. Herman

KrzysztofHerman, 2024-10-15:
@Ikram-rs22 Did it work for you ? Can we close the issue ?

Ikram-rs22, 2024-10-15:
Yes thank you. Please kindly close this issue

closed 2024-10-15 by KrzysztofHerman